realsenseai/librealsense #13514: Unhandled exception escaping from a worker thread!

dwu006, 2024-11-14:
| | |
|---|---|
|**librealsense**|2.56.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-11-15:
Hi @dwu006  The error that you experienced usually happens when attempting to access an incompatible camera (such as a laptop's built-in webcam) using the RealSense Viewer's 'Add Source' button.  The Add Source button can display the RGB of some non-RealSense webcams but not all.

Which RealSense camera model are you using with the 2.56.2 version of the Viewer, please?

dwu006, 2024-11-15:
Ah I realized i was trying to access my laptop-built camera. Thank you very much!

closed 2024-11-15 by dwu006

MartyG-RealSense, 2024-11-16:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!