realsenseai/librealsense #13516: Jetpack 6.0 support

PredyDaddy, 2024-11-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Yes, I read the doc said librealsense2.54.2 is introducing Jetpack 6.0 support, so now I am using 2.56.1 to use realsense in Jetpack 6.0

    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * I guess NO.
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.
    * I try to follow https://github.com/IntelRealSense/librealsense/issues/13326 this disscussion to install but not success. 

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i and L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.15.122-tegra                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I use follow command to install librealsense and I can successfully use realsense-viewer but when I connect my D435i and L515 and still cannot see my camera. 
```bash
sudo apt-get update
sudo apt-get install git cmake build-essential libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt-get install libcurl4-openssl-dev

git clone https://github.com/IntelRealSense/librealsense.git 
cd librealsense 
git checkout v2.56.1

mkdir build && cd build 
cmake .. -DBUILD_WITH_CUDA=ON -DBUILD_EXAMPLES=OFF -DCMAKE_BUILD_TYPE=Release

make -j$(nproc) 
sudo make install

sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
sudo udevadm control --reload-rules && sudo udevadm trigger
```
Can anyone help me look at it?

MartyG-RealSense, 2024-11-15:
Hi @PredyDaddy  The L515 camera will not work with librealsense 2.56.1 as support for it was removed in version 2.54.1 as it is a retired camera model.  So 2.53.1 is the final SDK / Viewer version that will detect an L515.

In regard to D435i, please try installing librealsense with the **RSUSB backend** installation method below.

```
git clone https://github.com/IntelRealSense/librealsense.git 
cd librealsense 
git checkout v2.56.1

mkdir build && cd build 
cmake .. -DFORCE_RSUSB_BACKEND=TRUE -DBUILD_WITH_CUDA=ON -DBUILD_EXAMPLES=OFF -DCMAKE_BUILD_TYPE=Release

make -j$(nproc) 
sudo make install
```

With the RSUSB method you do not need to sudo apt-install the dependencies or install the udev rules.  RSUSB = true bypasses the kernel and so is not dependent on a particular Linux version or kernel version and does not require a kernel patch script to be applied.

PredyDaddy, 2024-11-16:
hi, I can install realsense-sdk with following command in jetpack 6.0
```bash
git clone https://github.com/IntelRealSense/librealsense.git 
cd librealsense 
git checkout v2.56.1

mkdir build && cd build 
cmake .. -DFORCE_RSUSB_BACKEND=TRUE -DBUILD_WITH_CUDA=ON -DBUILD_EXAMPLES=OFF -DCMAKE_BUILD_TYPE=Release

make -j$(nproc) 
sudo make install
```
But I cannot use pyrealsense to use camera in python script. I guess because realsense2 version is `pyrealsense2 2.55.1.6486` so can I have other way to use pyrealsense?

MartyG-RealSense, 2024-11-16:
You can build the pyrealsense2 wrapper at the same time as librealsense if you add the flag below to the CMake build instruction.

`-DBUILD_PYTHON_BINDINGS:bool=true`

PredyDaddy, 2024-11-20:
New question, @MartyG-RealSense  
Cause my project need cuda 12.6 so I have to use realsense 2.56.1, but then I cannot use L515, I try D455 and D435, with really bad quality of pointcloud. Can I know how to use L515 in Jetpack 6.0?

MartyG-RealSense, 2024-11-20:
The obly way to use L515 would be to use librealsense 2.53.1,because when support for a camera is removed from a particular version of the librealsense SDK, all the code that makes that camera able to work is removed.  So there is no way to adjust a recent version such as 2.56.1 to work with an L515.

You should be able to use L515 with a JetPack 6-equipped Jetson if you build the librealsense SDK with the RSUSB BAckend method described earlier at https://github.com/IntelRealSense/librealsense/issues/13516#issuecomment-2478473878 though then you would still have the problem of the pyrealsense2 wrapper not working.

You would not necessarily need CUDA support in the SDK, as the SDK's CUDa support only enhances three particular types of function - depth-color alignment, pointcloud generation and YUY to RGB color conversion.  If you are not doing pointclouds or depth-color alignment then having the SDK's CUDA support enabled will not be beneficial to you as it will have no effect on performance.

SnoopyGeeek, 2024-11-25:
> Hi @PredyDaddy The L515 camera will not work with librealsense 2.56.1 as support for it was removed in version 2.54.1 as it is a retired camera model. So 2.53.1 is the final SDK / Viewer version [...]

We followed this instruction to install librealsense, but we cannot detect our any realsense devices. Our system is Jetson Orin Agx 64G with Ubuntu 22.04 and Jetpack 6.0.  The strange thing is that _lsusb_ command can detect but realsense-viewer and rs-enumerate-devices cannot. Here is our result running _lsusb_, _realsense-viewer_ and rs-enumerate-devices.
lsusb result: [image]
rs-enumerate-devices result:
[image]
realsense-viewer result:  
[image]

We got stuck with the problem for days. Could you plz help us out?

MartyG-RealSense, 2024-11-25:
@SnoopyGeeek You are very welcome.  Thanks very much!

MartyG-RealSense, 2024-11-25:
@PredyDaddy  As the creator of this issue, do you require further assistance please?  Thanks!

PredyDaddy, 2024-11-26:
> @PredyDaddy As the creator of this issue, do you require further assistance please? Thanks!作为此问题的创建者，您需要进一步的帮助吗？谢谢！

No, thank you very much!!!!

closed 2024-11-26 by PredyDaddy

MartyG-RealSense, 2024-11-26:
You are very welcome, @PredyDaddy - I'm pleased that I could help.  Thanks very much for the update!