realsenseai/librealsense #13517: SR305 Camera Detected by Windows, But Not by RealSense Viewer (Windows 10)

mstarckgh, 2024-11-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | v2.56.2.7411| 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi there,

I'm experiencing an issue with the Intel RealSense SR305 on Windows 10. While the camera is detected by Windows and listed in the Device Manager, and I can use it as a webcam, the RealSense Viewer does not recognize the camera and prompts me to "connect a RealSense camera."

Here’s a summary of the troubleshooting steps I’ve tried so far:
On Windows:

    Installed the RealSense SDK via the Windows .exe installer.
    Tested with different USB ports.
    Used a powered USB hub.
    Tried on a different computer.

On Linux (Ubuntu):

    The camera appears in the output of lsusb.
    Installed the RealSense SDK using the .deb package.
    Built the SDK from source.
    Tried different Linux distributions.

Unfortunately, none of these steps have resolved the issue, and I’m starting to run out of ideas. I’m not sure if this is a new issue or if I’m overlooking something simple.

Any help or suggestions would be greatly appreciated!

Kind regards,
Matthias

Kind regards
Matthias

MartyG-RealSense, 2024-11-16:
Hi @mstarckgh  Support for the SR300 / SR305 cameras was removed in RealSense SDK / Viewer version 2.51.1 as it is a retired model.  If you install the 2.50.0 version of the .exe then the camera should be detected in the Viewer. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

mstarckgh, 2024-11-16:
Hi,
Thanks for the quick response! Could you please clarify how long you plan to support the D455 and D435i models?

MartyG-RealSense, 2024-11-16:
There are no current plans to retire 400 Series models.  When models are retired, it is announced by Intel months in advance in End of Life (EOL) notices.

When a model is retired from sale, there is usually an extended period of time where it continues to be supported in the RealSense SDK and Viewer software.

closed 2024-11-16 by mstarckgh