realsenseai/librealsense #13521: Possible issue with patch-realsense-ubuntu-lts-hwe.sh

Mathlete-101, 2024-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | actually not relevant | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    |  e.g. 5.15.0-1063-raspi                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  lastest git commit as of 11/18/2024 }                          | 
| Language                            |  {n/a }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I'm trying to build from source, but when I did so, the build script [](https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-lts-hwe.sh) failed on line 178. The last two lines printed were as follows:
`Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.'

To try to diagnose this, I get on to the github and look at the script. Eventually, I check this history, and what do you know? The very last commit to this file on September 8 modified this exact section of code to work differently.

Mathlete-101, 2024-11-19:
Upon further analysis of the script, I don't know what's happening anymore. I'll reopen if there is a _real_ problem.

Mathlete-101, 2024-11-19:
I have determined that the real issue is, why is this version not supported or is there an issue with my os configuration?

MartyG-RealSense, 2024-11-20:
I reviewed the case again from the beginning.  As you were attempting to apply the hwe patch script to a kernel on a Raspberry Pi, that likely explains your initial problems with the **fatal: You need to specify a tag name** message, as this patch script is intended for use on desktop and laptop PCs with x86 / x64 architecture rather than Arm architecture devices such as Pi.

If you are able to use CMake then instead of using the libuvc_installation.sh script, a similar installation method described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 called **RSUSB Backend** can be used.  Like libuvc backend, RSUSB does not require a kernel patch script to be applied.

Mathlete-101, 2024-12-17:
I tried to install according to the rsusb backend method I got an error about a missing documentation file in libcurl when running make -j8. Most of the pieces were able to install except for realsense-viewer and rs-depth-quality. I thought that those might be enough to use the api but I installed the python wrapper and the camera didn't seem to be able to be detected.

MartyG-RealSense, 2024-12-18:
You can build the Python wrapper at the same time as librealsense by adding **-DBUILD_PYTHON_BINDINGS:bool=true** to the CMake build command.  Is this the method that you used to build the wrapper, please?

closed 2025-01-20 by MartyG-RealSense