realsenseai/librealsense #13530: Intel RealSense Touchless Control Software (TCS)

raf620, 2024-11-20:
| | |
|---|---|
|**librealsense**|2.56.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25602, json_settings:{):
    failed to convert special folder: errno=42

```

MartyG-RealSense, 2024-11-20:
Hi @raf620  Please see https://github.com/IntelRealSense/librealsense/issues/13523 for information about the errno=42 error.

closed 2024-12-01 by MartyG-RealSense

MartyG-RealSense, 2024-12-04:
My Intel RealSense colleagues have provided some further advice about the errno=42 error.

****************

Given that the problem seems to be related to certain charcters being used in a Windows user account name, I asked whether creating a secondary Windows user account with a name that has no special characters might help.  This was their response. 

You can try to run it as Administrator instead of switching the account.  On Windows 11, it also has a feature to run the app as a different user.  That option can be enabled with the instructions at the link below:
 
https://winaero.com/how-to-run-apps-as-different-user-in-windows-11/

raf620, 2024-12-04:
Hello!  Thank you very much, for the help I will check Thank you