realsenseai/librealsense #13534: D450 depth module and Vision processor D4 V3 not detected by Windows or Ubuntu

Ojas-Chanakya131, 2024-11-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D450| 
| Operating System & Version |   {Win 10 / Ubuntu 22.04 | 
| Platform                                 | PC/ NVIDIA Jetson |
| SDK Version                            | 2.5.5.1| 
| Segment			| Robot | 

### Issue Description
Hello,
We have bought 5 d450 depth modules, vision processor d4 v3s and rigid interposers. Whenever I connect the board to the laptop or jetson, in Windows or Ubuntu, it is never able to get any camera feed. I have tried realsense-ros, realsense viewer in both windows and ubuntu, none seem to be able to recognize it. When i connect just the vision processor it shows it as a tester so i guess it is working. In windows it shows that the device has malfunctioned.

Any help would be appreciated
Thank you

MartyG-RealSense, 2024-11-22:
Hi @Ojas-Chanakya131  It sounds as though the rigid interposer might not be functioning.  The camera identifies as D405i Tester when the depth module is not connected to the interposer, so only the Vision Processor D4 board is activated.

Have you connected the D450 to the 50-pin connector on top of the long side of the board and not the connector on the short side?

[image]

MartyG-RealSense, 2024-12-01:
Hi @Ojas-Chanakya131  Do you require further assistance with this case, please?  Thanks!

Ojas-Chanakya131, 2024-12-01:
Hello,
I apologise for the late reply, I had an accident and had to take a few
days off work. Regardless, I did try what you said and even after using a
different interposer, and different depth module and board and trying
permutations and combinations of those, it still doesn't work. I did search
online and I saw a lot of people having trouble with cables. So could it
just be a bad batch of interposers or are there any specific cables I
should look into?

MartyG-RealSense, 2024-12-01:
If the interposer is not at fault then this situation can also occur because of a version mismatch between the D450 board and the Vision Processor D4 V3 board.  There are different versions of the D450 module, ' K83121-100' and ' K83121-101'.  If the purchased D450 module is the newer '101' version then it may not work with the D4 V3 board if it has an old firmware version pre-installed on it.

The D450 module should have a sticker on it that will say whether it is the K83121-101 module.  If you do have the -101 model and you purchased the D450 and D4 V3 separately instead of a paired D450+D4 multipack (which ensures compatibility) then they may be mismatched.

A single flexible cable version of the interposer can be purchased from the RealSense Store to confirm or eliminate whether the rigid interposer is responsible.

If the problem is with a version mismatch due to an old firmware pre-installed on the D4 V3 boards then this cannot be fixed by the end-user.  If you do have the -101 version of the D450 board then I will discuss it with my Intel RealSense colleagues.

MartyG-RealSense, 2024-12-01:
A single unit of the flexible interposer cable can be purchased from the link below.

https://store.intelrealsense.com/buy-hsf100-flexi-interposer-cable.html

Ojas-Chanakya131, 2024-12-01:
Hello,
I will check out the version and update you regarding the same. If there is
a version mismatch, will I have to replace the camera modules, or the
vision board?

Thank you for your support

MartyG-RealSense, 2024-12-01:
If there is a version mismatch, I will consult with my colleagues about what the next recommended step should be.  Only the Vision Processor D4 V3 board would need to be firmware-updated or replaced, as the firmware is stored on the D4 board.  Nothing would have to be done to the D450 modules.  If the recommended advice was to update the existing D4 boards then they would have to be mailed to Intel for the procedure to be carried out and then be shipped back to you.

In case it comes to that, may I ask what country you are located in please?

Ojas-Chanakya131, 2024-12-01:
I am located in India

MartyG-RealSense, 2024-12-01:
Okay, thanks very much.

Please let me know if your D450 modules are the K83121-101 version when you have the opportunity to check and then we can take it from there.  Thanks!

Ojas-Chanakya131, 2024-12-01:
Thanks for the help!, it's still Sunday here so I will let you know when I
go into the office tomorrow

MartyG-RealSense, 2024-12-03:
Hi again, please email me at the address below to further discuss shipping related matters for your five D4 V3 boards.

martyx.grover@intel.com

MartyG-RealSense, 2025-01-06:
Case closed due to resolution by email.

closed 2025-01-06 by MartyG-RealSense