realsenseai/librealsense #13536: error install  librealsense2-dkms

lianghongzhuo, 2024-11-22:
Hi,
I tried to install ` librealsense2-dkms` using the command `apt install librealsense2-dkms` in a ubuntu22.04 docker image.
I try to install it as this package is listed in the install instructions.
And get the following error.
However, the realsense can be used normally. 
My questions are:
1. Is this package necessary? If so, in what situation?
2. What's the reason that made this command fail?
Thank you!

```
apt install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  cpp-12 dctrl-tools dkms fakeroot gcc-12 kmod libasan8 libfakeroot libgcc-12-dev libtsan2 linux-headers-5.15.0-126 linux-headers-5.15.0-126-generic
  linux-headers-generic
Suggested packages:
  gcc-12-locales cpp-12-doc debtags menu gcc-12-multilib gcc-12-doc
The following NEW packages will be installed:
  cpp-12 dctrl-tools dkms fakeroot gcc-12 kmod libasan8 libfakeroot libgcc-12-dev librealsense2-dkms libtsan2 linux-headers-5.15.0-126
  linux-headers-5.15.0-126-generic linux-headers-generic
0 upgraded, 14 newly installed, 0 to remove and 0 not upgraded.
Need to get 58.6 MB of archives.
After this operation, 256 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://archive.ubuntu.com/ubuntu jammy/main amd64 kmod amd64 29-1ubuntu1 [101 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 cpp-12 amd64 12.3.0-1ubuntu1~22.04 [10.8 MB]
Get:3 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2-dkms all 1.3.28-0ubuntu1 [2,911 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 libasan8 amd64 12.3.0-1ubuntu1~22.04 [2,442 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 libtsan2 amd64 12.3.0-1ubuntu1~22.04 [2,477 kB]
Get:6 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 libgcc-12-dev amd64 12.3.0-1ubuntu1~22.04 [2,618 kB]
Get:7 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 gcc-12 amd64 12.3.0-1ubuntu1~22.04 [21.7 MB]
Get:8 http://archive.ubuntu.com/ubuntu jammy/main amd64 dctrl-tools amd64 2.24-3build2 [66.9 kB]
Get:9 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 dkms all 2.8.7-2ubuntu2.2 [70.1 kB]
Get:10 http://archive.ubuntu.com/ubuntu jammy/main amd64 libfakeroot amd64 1.28-1ubuntu1 [31.5 kB]
Get:11 http://archive.ubuntu.com/ubuntu jammy/main amd64 fakeroot amd64 1.28-1ubuntu1 [60.4 kB]
Get:12 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-5.15.0-126 all 5.15.0-126.136 [12.3 MB]
Get:13 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-5.15.0-126-generic amd64 5.15.0-126.136 [2,868 kB]
Get:14 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-headers-generic amd64 5.15.0.126.125 [2,346 B]
Fetched 58.6 MB in 4s (16.3 MB/s)                            
Selecting previously unselected package kmod.
(Reading database ... 104121 files and directories currently installed.)
Preparing to unpack .../00-kmod_29-1ubuntu1_amd64.deb ...
Unpacking kmod (29-1ubuntu1) ...
Selecting previously unselected package cpp-12.
Preparing to unpack .../01-cpp-12_12.3.0-1ubuntu1~22.04_amd64.deb ...
Unpacking cpp-12 (12.3.0-1ubuntu1~22.04) ...
Selecting previously unselected package libasan8:amd64.
Preparing to unpack .../02-libasan8_12.3.0-1ubuntu1~22.04_amd64.deb ...
Unpacking libasan8:amd64 (12.3.0-1ubuntu1~22.04) ...
Selecting previously unselected package libtsan2:amd64.
Preparing to unpack .../03-libtsan2_12.3.0-1ubuntu1~22.04_amd64.deb ...
Unpacking libtsan2:amd64 (12.3.0-1ubuntu1~22.04) ...
Selecting previously unselected package libgcc-12-dev:amd64.
Preparing to unpack .../04-libgcc-12-dev_12.3.0-1ubuntu1~22.04_amd64.deb ...
Unpacking libgcc-12-dev:amd64 (12.3.0-1ubuntu1~22.04) ...
Selecting previously unselected package gcc-12.
Preparing to unpack .../05-gcc-12_12.3.0-1ubuntu1~22.04_amd64.deb ...
Unpacking gcc-12 (12.3.0-1ubuntu1~22.04) ...
Selecting previously unselected package dctrl-tools.
Preparing to unpack .../06-dctrl-tools_2.24-3build2_amd64.deb ...
Unpacking dctrl-tools (2.24-3build2) ...
Selecting previously unselected package dkms.
Preparing to unpack .../07-dkms_2.8.7-2ubuntu2.2_all.deb ...
Unpacking dkms (2.8.7-2ubuntu2.2) ...
Selecting previously unselected package libfakeroot:amd64.
Preparing to unpack .../08-libfakeroot_1.28-1ubuntu1_amd64.deb ...
Unpacking libfakeroot:amd64 (1.28-1ubuntu1) ...
Selecting previously unselected package fakeroot.
Preparing to unpack .../09-fakeroot_1.28-1ubuntu1_amd64.deb ...
Unpacking fakeroot (1.28-1ubuntu1) ...
Selecting previously unselected package linux-headers-5.15.0-126.
Preparing to unpack .../10-linux-headers-5.15.0-126_5.15.0-126.136_all.deb ...
Unpacking linux-headers-5.15.0-126 (5.15.0-126.136) ...
Selecting previously unselected package linux-headers-5.15.0-126-generic.
Preparing to unpack .../11-linux-headers-5.15.0-126-generic_5.15.0-126.136_amd64.deb ...
Unpacking linux-headers-5.15.0-126-generic (5.15.0-126.136) ...
Selecting previously unselected package linux-headers-generic.
Preparing to unpack .../12-linux-headers-generic_5.15.0.126.125_amd64.deb ...
Unpacking linux-headers-generic (5.15.0.126.125) ...
Selecting previously unselected package librealsense2-dkms.
Preparing to unpack .../13-librealsense2-dkms_1.3.28-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.28-0ubuntu1) ...
Setting up cpp-12 (12.3.0-1ubuntu1~22.04) ...
Setting up linux-headers-5.15.0-126 (5.15.0-126.136) ...
Setting up linux-headers-5.15.0-126-generic (5.15.0-126.136) ...
Setting up linux-headers-generic (5.15.0.126.125) ...
Setting up kmod (29-1ubuntu1) ...
Setting up libfakeroot:amd64 (1.28-1ubuntu1) ...
Setting up fakeroot (1.28-1ubuntu1) ...
update-alternatives: using /usr/bin/fakeroot-sysv to provide /usr/bin/fakeroot (fakeroot) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/man1/fakeroot.1.gz because associated file /usr/share/man/man1/fakeroot-sysv.1.gz (of link
 group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/man1/faked.1.gz because associated file /usr/share/man/man1/faked-sysv.1.gz (of link group
 fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/es/man1/fakeroot.1.gz because associated file /usr/share/man/es/man1/fakeroot-sysv.1.gz (o
f link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/es/man1/faked.1.gz because associated file /usr/share/man/es/man1/faked-sysv.1.gz (of link
 group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/fakeroot.1.gz because associated file /usr/share/man/fr/man1/fakeroot-sysv.1.gz (o
f link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/fr/man1/faked.1.gz because associated file /usr/share/man/fr/man1/faked-sysv.1.gz (of link
 group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/sv/man1/fakeroot.1.gz because associated file /usr/share/man/sv/man1/fakeroot-sysv.1.gz (o
f link group fakeroot) doesn't exist
update-alternatives: warning: skip creation of /usr/share/man/sv/man1/faked.1.gz because associated file /usr/share/man/sv/man1/faked-sysv.1.gz (of link
 group fakeroot) doesn't exist
Setting up libasan8:amd64 (12.3.0-1ubuntu1~22.04) ...
Setting up libtsan2:amd64 (12.3.0-1ubuntu1~22.04) ...
Setting up dctrl-tools (2.24-3build2) ...
Setting up libgcc-12-dev:amd64 (12.3.0-1ubuntu1~22.04) ...
Setting up gcc-12 (12.3.0-1ubuntu1~22.04) ...
Setting up dkms (2.8.7-2ubuntu2.2) ...
Setting up librealsense2-dkms (1.3.28-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.28 DKMS files...
It is likely that 6.2.0-39-generic belongs to a chroot's host
Building for 5.15.0-126-generic
Building initial module for 5.15.0-126-generic
Done.

uvcvideo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-126-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-126-generic/updates/dkms/

depmod...

Loading the modified modules into kernel...dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

MartyG-RealSense, 2024-11-23:
Hi @lianghongzhuo  The **librealsense2-dkms** package is vital when performing a package installation of the librealsense SDK on a PC computer, as it is the core of librealsense and the SDK cannot function without it.

It looks as though you are using the **distribution_linux.md** package installation instructions.  Is this correct, please?  If you are using these instructions, can you confirm if you are using a PC desktop or laptop computer?

 If an **Nvidia Jetson** board was being used as the computer instead of a PC then a different set of dkms package installation instructions at the link below should be used.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md

https://github.com/IntelRealSense/librealsense/issues/6627#issuecomment-647043443 advises that the messages **Running module version sanity check. - Original module - No original module exists within this kernel** may occur if a modified kernel is being used.

********

Instead of installing the SDK from packages, if it is built from source code with the **libuvc backend** installation method then it will bypass the kernel and reduce the likelihood of a kernel-related installation problem occurring.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend procedure uses a pre-made build script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

lianghongzhuo, 2024-11-23:
I am trying on both x86 desktop computer and x86 laptop. All failed in Docker.
And yes, the installation instruction is from: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
Any suggestions?

MartyG-RealSense, 2024-11-23:
As the advice in the link suggested, you could run the command `sudo apt dist-upgrade` to make sure that you have standard kernel modules instead of a possibly modified kernel.

If you do not think that you should modify the computer's current kernel version then I recommend installing the SDK from source code with the libuvc backend procedure, which should allow the SDK to work without you having to alter the computer's kernel.

lianghongzhuo, 2024-11-23:
One thing need to mention is that my computer has a higher kernel version installed:
```
 uname -a
Linux 087005c9ef2c 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```
and install librealsense2-dkms will leads a downgrade of the kernel:
```
The following additional packages will be installed:
  cpp-12 dctrl-tools dkms fakeroot gcc-12 kmod libasan8 libfakeroot libgcc-12-dev libtsan2 linux-headers-5.15.0-126 linux-headers-5.15.0-126-generic
  linux-headers-generic
Suggested packages:
  gcc-12-locales cpp-12-doc debtags menu gcc-12-multilib gcc-12-doc
The following NEW packages will be installed:
  cpp-12 dctrl-tools dkms fakeroot gcc-12 kmod libasan8 libfakeroot libgcc-12-dev librealsense2-dkms libtsan2 linux-headers-5.15.0-126
  linux-headers-5.15.0-126-generic linux-headers-generic
0 upgraded, 14 newly installed, 0 to remove and 0 not upgraded.

```

lianghongzhuo, 2024-11-23:
I found the solution:
before install `librealsense2-dkms`
I manually install `apt install linux-headers-6.2.0-39-generic`
then it works!
I think the reason is that `librealsense2-dkms` fails to find the correct linux-headers to install.
as show at first comment:  it try to install linux-headers-5.15.0-126-generic

MartyG-RealSense, 2024-11-23:
It's great to hear that you achieved a solution.   Thanks very much for the update and for sharing the details of what worked for you!

lianghongzhuo, 2024-11-23:
Is there a upstream solution to correctly select linux-headers?

lianghongzhuo, 2024-11-23:
I can also run `uname -r` to check the correct version to install before install librealsense2-dkms

closed 2024-11-23 by lianghongzhuo

flxinxout, 2025-02-05:
Hello,

I have exactly the same issue with the same docker image and trying to install the same package. I have on my host machine an Ubuntu 24.04 with linux headers 6.8.2-generic. I did something like this in my dockerfile

```
RUN mkdir -p /etc/apt/keyrings

# Install librealsense
RUN curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/keyrings/librealsense.pgp > /dev/null
RUN echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/librealsense.list
RUN apt-get update

RUN apt-get install linux-headers-6.8.0-52-generic -y

RUN apt-get update && apt-get install -y \
    librealsense2-dkms \
    librealsense2-utils
```

But I get exactly the same error of the initial post. Did someone add something else in the file?
Thanks

MartyG-RealSense, 2025-02-05:
Hi @flxinxout  Are you using kernel version **6.8.0-52-generic** please?  If you are then support for manually patching a source code build of kernel 6.8 was not added to the librealsense SDK until version 2.56.2 beta onwards.  

And the dkms packages - which contain the kernel patch within them and so do not require the manual application of a patch script - only list support up to kernel 6.5 at the time of writing this.

[image]

flxinxout, 2025-02-05:
Hello @MartyG-RealSense thanks for the quick reply. Yes it's my kernel version.
I found this post saying that kernel 6.8 was supported with ubuntu 24.04

https://github.com/IntelRealSense/librealsense/issues/13314#issuecomment-2337282509

MartyG-RealSense, 2025-02-06:
The comment refers to "manual patches for Kernel 6.8 Jammy and Noble".  Manual patching is the running of a script to patch the kernel when using a librealsense build that was compiled from source code.  

DKMS package installations do not use manual patching as the patch is bundled within the packages, but it means that if support for a particular kernel type (such as 6.8) is not listed then it will not be possible to use package installation if the computer is not using one of the supported kernels.

jackjansen, 2026-02-14:
I got to this issue while trying to install librealsense onto the Github `linux-latest` runners. But the install fails. Here are the relevant lines from the log:

```
2026-02-14T13:57:59.0759606Z Setting up librealsense2-dkms (1.3.28-0ubuntu1) ...
2026-02-14T13:57:59.1512704Z Loading new librealsense2-dkms-1.3.28 DKMS files...
2026-02-14T13:57:59.2665927Z Building for 6.14.0-1017-azure
2026-02-14T13:57:59.3035737Z Building initial module for 6.14.0-1017-azure
2026-02-14T13:57:59.6076307Z Secure Boot not enabled on this system.
2026-02-14T13:58:00.2592801Z Error! Bad return status for module build on kernel: 6.14.0-1017-azure (x86_64)
2026-02-14T13:58:00.2593537Z Consult /var/lib/dkms/librealsense2-dkms/1.3.28/build/make.log for more information.
2026-02-14T13:58:00.2656741Z dpkg: error processing package librealsense2-dkms (--configure):
2026-02-14T13:58:00.2657988Z  installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
```

How can I get librealsense onto a GitHub runner (for building my package that depends on it only, plus for running my testsuite but obviously only insofar as it doesn't require hardware).

MartyG-RealSense, 2026-02-14:
Hi @jackjansen  I am not familiar with GitHub Runners and so do not have advice to offer about that subject, unfortunately.

If you were trying to build librealsense through GitHub then typically the way to do that would be to use a **git clone** instruction.

`git clone https://github.com/realsenseai/librealsense.git`