realsenseai/librealsense #13539: 3588

lian6605, 2024-11-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

lian6605, 2024-11-25:
I am using rk3588 from boardcon.
Ubuntu 20.04 
kernel 5.10.209
According to the official website 
Ubuntu 18/20 with LTS kernel (< 5.13)
$./scripts/patch-realsense-ubuntu-lts.sh
I did, but, 
Unsupported kernel version 5.10.209 . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
I did it.

Reading package lists... Done
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal/universe/binary-i386/Packages  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports/dists/focal/main/binary-i386/Packages  404  Not Found [IP: 101.6.15.130 80]
E: Failed to fetch http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports/dists/focal-updates/main/binary-i386/Packages  404  Not Found [IP: 101.6.15.130 80]
E: Failed to fetch http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports/dists/focal-backports/main/binary-i386/Packages  404  Not Found [IP: 101.6.15.130 80]
E: Failed to fetch http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports/dists/focal-security/main/binary-i386/Packages  404  Not Found [IP: 101.6.15.130 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
root@rk3588s-ubuntu:~/download/librealsense# rs-enumerate-devices
bash: rs-enumerate-devices: command not found
root@rk3588s-ubuntu:~/download/librealsense# rs-enumerate-devices
bash: rs-enumerate-devices: command not found

MartyG-RealSense, 2024-11-25:
Hi @lian6605  For RK (Rockchip) devices it is recommended to install the librealsense SDK from source code using the **libuvc backend** method at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

A libuvc backend build of librealsense bypasses the Linux kernel and so is not dependent on a particular Linux version or kernel version and does not need a kernel patch script to be applied.

The libuvc method provides a pre-made build script called 'libuvc_installation.sh'.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

closed 2024-12-07 by MartyG-RealSense