realsenseai/librealsense #13540: Kinfu‘s performance is poor.

star-ice, 2024-11-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux(20.04) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NIVDIA Jetson Orin NX |
| SDK Version                            |  | 
| Language                            | C++  | 
| Segment			|  Robot  | 

### Issue Description
I used Kinfu for the 3D reconstruction of a small scene.
The camera is D455.
I opened the rs-kinfu program and slowly moved the camera, but the performance was not good enough.
The point cloud was skewed and even though I had moved as slowly as I could, there were still so many redundant points around the object.
The point cloud is as follows:
[image]
The scene looks like this:
[image]

I wonder whether I used kinfu incorrectly or need to modify the code for better performance.
Thanks for your help.

MartyG-RealSense, 2024-11-25:
Hi @star-ice  It sounds as though you used rs-kinfu properly.  Other RealSense users have experienced slow performance with rs-kinfu, so it is not just you.

I would recommend trying the following changes:

1.  Comment out line 204 defining the **high_density** camera configuration preset.

2.  Comment out lines 219 and 257-259 that apply post-processing filters, as these filters are calculated on the computer's CPU and so can add a processing burden to it.  If commenting out all the filters results in a noticable reduction in depth image quality then try only commenting out the spatial filter.

star-ice, 2024-11-26:
It looks quite normal now. Thanks.
I still have another question. Can I use bag file as the input for offline reconstruction? Is there any suggestion for modifing the code?

MartyG-RealSense, 2024-11-26:
Yes, you can use a bag file as the data source for a RealSense program script instead of a live camera.  You can do this in C++ by inserting a **cfg.enable_device_from_file()** instruction immediately before the pipe start line.

```
rs2::config cfg;
cfg.enable_device_from_file(<filename>);
pipe.start(cfg); // Load from file
```

star-ice, 2024-11-27:
Thank you very much @MartyG-RealSense .

star-ice, 2024-12-02:
I have another question about building rs-kinfu. Every time I modify the code, I need to build the whole project via CMake and make, which takes time on Jetson. Can I only build rs-kinfu module?

MartyG-RealSense, 2024-12-02:
https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237 advises that only the modified example will be built when **make** is run and not the entire librealsense SDK.

I believe that it is advising not to repeat the `cmake ../` command and just input the `make` command that comes after that line whilst in the librealsense 'build' folder.

star-ice, 2024-12-03:
> #2946 (comment)

Thank you. It helps a lot.

MartyG-RealSense, 2024-12-03:
You are very welcome!

star-ice, 2024-12-12:
Hi, there is another question also related to the performance.

I have a set of objects to be modelled in 3D, including a box, orange, coca-cola, tape, bottled water, etc. The camera was set on a robotic arm to ensure smooth movement when capturing frames. The scan trajectory is nearly a circle above(around 60cm) the object. 

I found that rs-kinfu only performed well on the box. Coca-cola, tape, and bottled water were almost disappeared in the pcd. 

The rgbd images of some objects are in the below:
[image]
[image]
[image]

And the results of rs-kinfu are:
[image]
[image]
[image]

Is there any advice for improving rs-kinfu?

MartyG-RealSense, 2024-12-12:
You could edit line 204 of rs-kinfu to change the camera configuration preset that is applied to the depth data.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L204

If you try different configurations then you may find one that suits these non-box objects.  A table at the link below describes the benefits that each preset provides.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

So on line 204, you would change the **RS2_RS400_VISUAL_PRESET_HIGH_DENSITY** part of the line to one of the following:

 RS2_RS400_VISUAL_PRESET_CUSTOM
 RS2_RS400_VISUAL_PRESET_DEFAULT
 RS2_RS400_VISUAL_PRESET_HAND
 RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY
 RS2_RS400_VISUAL_PRESET_HIGH_DENSITY
 RS2_RS400_VISUAL_PRESET_MEDIUM_DENSITY

'Custom' is the default preset of the RealSense Viewer.  So it may be worth trying that one first if you got good results with the objects in the Viewer.

Medium_Density is also a preset that can often improve images, as it provides a good balance between accuracy and the amount of detail on the depth image.

star-ice, 2024-12-13:
Thanks for your reply. I will try it. The input of kinfu is a bag file and I may not change the camera settings with it but I will try to set it when I'm using the camera.

Yesterday I tried to change the parameter {max_dist} at line 16. I found that the empty points were reduced when I set the value from 1.0 to 6.0. I also commented some lines related to visualization, e.g. `window app(1280, 720, "RealSense KinectFusion Example");`, for saving memory. The performance was also better than before.

I don't know why those changes gave me better results or whether they made sense.

MartyG-RealSense, 2024-12-13:
Thanks very much for the update.  It's great to hear that you achieved improvement.

**max_dist** will determine the maximum depth value in meters that can be rendered on the depth image.  So it makes sense that more of the detail would be filled in if the image is able to include depth points that are greater than 1 meter away (up to 6 meters after your change).

MartyG-RealSense, 2025-01-06:
Hi @star-ice  Do you require further assistance with this case, please?  Thanks!

star-ice, 2025-01-07:
> Hi @star-ice Do you require further assistance with this case, please? Thanks!

No. Thank you.

MartyG-RealSense, 2025-01-07:
Okay, thanks very much for the update!

closed 2025-01-07 by MartyG-RealSense

star-ice, 2025-01-09:
Hi, I still have another question. The documentation of rs-kinfu mentioned that kinfu uses GPU but when I ran the program, all CPUs were used. I've checked the Cuda env about my jetson and it was normal. Do I need to add some env settings for program to use GPU?

MartyG-RealSense, 2025-01-09:
Looking at the code of rs-kinfu, it appears that its 'frame fusion' process generates a pointcloud in an unconventional way that will not be GPU-accelerated by librealsense's CUDA support on Jetson boards.