realsenseai/librealsense #13542: How to convert uint8 data captured by Depth to the correct 'mono16'？

wenmingxiaohuo, 2024-11-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-11-25:
Hi @wenmingxiaohuo  RealSense depth data is not 8-bit.  It is 16-bit **uint16_t**.

wenmingxiaohuo, 2024-11-25:
`filename='D:\RecordData_D435\20241125_111919.bag';
% validateattributes(filename, {'char','string'}, {'scalartext', 'nonempty'}, '', 'filename', 1);
bag=rosbag(filename);
mes=select(bag,'Topic','/device_0/sensor_0/Depth_0/image/data');
poseStructs = readMessages(mes, 'DataFormat', 'struct'); %`
Getting information from the depth sensor，
[image]
The Data is unit8 but the decoding method is mono16.

wenmingxiaohuo, 2024-11-25:
Thanks, but I'm still a little confused.
I want to convert 814080 of uint8 to 407040 of uint16, what should I do?

wenmingxiaohuo, 2024-11-25:
Ok thanks for the answer, I see what you mean. Also in response to your reply there is a small question, 'D:\RecordData_D435\20241125_111919.bag' is the data captured using the D435, how can I tell if this bag file is a ROG bag file?

wenmingxiaohuo, 2024-11-25:
Sorry, I have other things to take care of, so please just leave a message at your leisure. Thank you for every answer, bye!

MartyG-RealSense, 2024-11-25:
A quick way to test whether a bag file might be a ROS bag is to drag and drop the file into the center panel of the RealSense Viewer tool if you have access to that tool.  If there is an error message instead of the bag starting to play automatically then there is a good chance that it is a rosbag instead of an SDK-bag.

wenmingxiaohuo, 2024-11-29:
I'd like to reiterate my question.
Recorded packets show depth data is a mono16 form, that is, it should be a layer of uint16 data, however, the actual data points are 1280 * 720 twice the amount, and the form is uint8, it should be reasonably to 2 for a group to uint16 format, how to realize it?
[image]

MartyG-RealSense, 2024-11-29:
I do not have knowledge about this particular subject.  It may be worth looking at the librealsense SDK's **/sensor_msgs/image_encodings.h** source code file and doing a search on the page for the word **mono16** to find relevant references in the code.

https://github.com/IntelRealSense/librealsense/blob/master/third-party/realsense-file/rosbag/msgs/sensor_msgs/image_encodings.h

closed 2024-12-13 by MartyG-RealSense