realsenseai/librealsense #13544: where is the C++  API of "Golden reset"?

diplomatist, 2024-11-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (5.16.0.1) | 
| Operating System & Version |   {Ubuntu20.04 focal LTS  | 
| Kernel Version (Linux Only)    |  (5.10.161)                                         | 
| Platform                                 | rk3588s |
| SDK Version                            |  { 2.56.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello, I would like to perform a "Golden reset" on my camera. Apart from resetting it through the calibrateData of realsense viewer/depth quality, are there any C++APIs or sample programs available for resetting?

At present, I can only control the camera through SSH and cannot directly run the realsense viewer/depth quality program as conveniently as on Windows.

diplomatist, 2024-11-25:
I found a C++interface with the following link, which should be open for use, right?
[rs.cpp_line3176](https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/src/rs.cpp#L3176)

[image]

MartyG-RealSense, 2024-11-25:
Hi @diplomatist  Yes, the RealSense SDK (librealsense) is open-source and you are free to use and modify its code.

diplomatist, 2024-11-25:
thanks，I will test rs2_reset_to_factory_calibration() api.

closed 2024-12-07 by MartyG-RealSense