realsenseai/librealsense #13545: Assuming a rate of 30FPS, a resolution of 1280*720, and an acquisition of 5 seconds, how is the size of the .bag file calculated?

wenmingxiaohuo, 2024-11-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

wenmingxiaohuo, 2024-11-26:
Assuming a rate of 30FPS, a resolution of 1280*720, and an acquisition of 5 seconds, how is the size of the .bag file calculated?

MartyG-RealSense, 2024-11-26:
Hi @wenmingxiaohuo  There is not a formula for calculating the size of a bag file in advance.  However, I recorded a depth stream of 1280x720 at 30 FPS to a bag file for 5 seconds and the file was 143 mb in size.

wenmingxiaohuo, 2024-11-26:
Thank you very much for your reply. I don't think the depth stream data is the whole content of the .bag file, although there is no specific formula to calculate it, but I think I should be able to calculate the size of the .bag file if I accumulate the whole data of the .bag. Suppose the RGB data rate is 30fps, resolution 480*640

wenmingxiaohuo, 2024-11-26:
Thank you very much for your reply. I don't think the depth stream data is the whole content of the .bag file, although there is no specific formula to calculate it, but I think I should be able to calculate the size of the .bag file if I accumulate all the data of the .bag. How should I calculate it? Suppose the RGB data rate is 30fps, resolution 480*640

MartyG-RealSense, 2024-11-26:
I recorded another 5 second, 30 FPS test bag with depth 1280x720 and RGB 640x480 and the bag was 267 mb.

wenmingxiaohuo, 2024-11-26:
Your work is recognizable and commendable, thank you for the test!

closed 2024-12-13 by MartyG-RealSense