realsenseai/librealsense #13547: Hello professionals, how can I turn on the camera through matlab?

wenmingxiaohuo, 2024-11-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

wenmingxiaohuo, 2024-11-27:
Hello professionals, how can I turn on the D435 camera via matlab?

wenmingxiaohuo, 2024-11-28:
Unlike #13546, I just want to start the D435 via matlab and have it work properly and continuously, not get a frame, can you help me with this?

MartyG-RealSense, 2024-11-28:
If you want to stream with the RealSense MATLAB wrapper instead of getting one frame then https://github.com/IntelRealSense/librealsense/issues/2705 and https://github.com/IntelRealSense/librealsense/issues/5514 may be helpful discussions.

There is also a discussion at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360038629154-How-to-stream-video-from-a-d435-camera-in-matlab

closed 2024-12-13 by MartyG-RealSense