realsenseai/librealsense #13550: matlab pointcloud

lrl-hub, 2024-11-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11)   | 
| SDK Version                            |  2.54.1             | 
| Language                            |  matlab   | 

I'm trying to figure out if it's possible to obtain the point cloud of a specific object using MATLAB.
In the MATLAB toolkit provided by the SDK, I can get the real-time point cloud of the camera using "pointcloud_example", but I can't obtain the complete point cloud of a specific object (for example, a sphere, the point cloud in the shape of a sphere). I'd like to ask if there is such a MATLAB program that can achieve this.

MartyG-RealSense, 2024-11-27:
Hi @lrl-hub  If you want an object-shaped pointcloud that only has depth detail of the specific object then an easy way to do that may be to put a non-glossy black background behind / under the object.  Then the camera should not be able to read depth detail from the black surface and it will appear on the pointcloud as an empty black area around the object.

lrl-hub, 2024-11-27:
OK, thank you. I've got a general understanding. I have another question. When using "pointcloud_example" in MATLAB, only the point cloud of the real-time camera can be obtained. For example, for a sphere, only the point cloud of one side of it can be obtained, and the point cloud of the invisible part cannot be obtained. Is there a solution to this?

MartyG-RealSense, 2024-11-27:
Obtaining a point cloud of a sphere would be like obtaining one of the human face.  From the front you can capture the face and a bit of the ears but not the sides of the head because the camera cannot see that detail from the front-on, straight ahead position.  To obtain a 360 degree pointcloud you would have to take at least 2 captures and stitch them together.

The RealSense SDK provides a C++ tutorial example for doing this, using MATLAB to obtain calibration information and feed it into the stitching program.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pointcloud/pointcloud-stitching

MartyG-RealSense, 2024-11-27:
If you can only use a single camera then you will likely have to export different views to multiple .ply pointcloud data files and then stitch them together in MATLAB using **ICP registration**, as described in the multiple-ply stitching discussion at the link below.

https://uk.mathworks.com/matlabcentral/answers/259918-merging-point-clouds-to-get-a-3-d-model

MartyG-RealSense, 2024-12-13:
Case closed due to no further comments received.

lrl-hub, 2025-01-03:
I want to use the RealSense D455 camera to create a dataset of a deformed balloon in order to obtain the 3D real points of the deformed balloon and the corresponding 2D points in the image. The total number of images is about 8, and the number of real points in each image is about 100. The obtained data will be used in MATLAB. What should I do?

MartyG-RealSense, 2025-01-03:
Once you have stitched a pointcloud together within MATLAB using a set of .ply files created in RealSense Viewer of different views of the balloon and stitched them together with ICP - suggested above at https://github.com/IntelRealSense/librealsense/issues/13550#issuecomment-2503724376 - then you could possibly retrieve the XYZ points of the cloud from within MATLAB.  Googling for the term **matlab pointcloud get xyz points** should provide some useful references in the search results about how to do so, like the links below.

https://uk.mathworks.com/matlabcentral/answers/1703500-extracting-xyzpoints-from-point-cloud
https://uk.mathworks.com/help/ros/ref/readxyz.html

lrl-hub, 2025-01-04:
Thank you for your reply. I'll give it a try.

lrl-hub, 2025-01-04:
I can use the following Matlab code to obtain the coordinate information of 100 randomly selected points in a frame of image. Is there a way to obtain the point cloud information of 100 non-random points? For example, if I want to obtain the point cloud information of 100 points for each of 4 frames of images, ensuring that these 100 points are the same points in these 4 frames of images.

function p100point()
    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();

    pointcloud = realsense.pointcloud();

    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();

    try
        % Obtain frames from a streaming device
        fs = pipe.wait_for_frames();

        % Select depth frame
        depth = fs.get_depth_frame();

        % Produce pointcloud
        if (depth.logical())
            points = pointcloud.calculate(depth);

            % Adjust frame CS to matlab CS
            vertices = points.get_vertices();
            X = vertices(:, 1, 1);
            Y = vertices(:, 2, 1);
            Z = vertices(:, 3, 1);

            % 随机选择100个点
            idx = randperm(length(X), 100);
            randomPointsX = X(idx);
            randomPointsY = Y(idx);
            randomPointsZ = Z(idx);

            % 可视化100个点
            plot3(randomPointsX, randomPointsZ, -randomPointsY, 'ro');
            grid on;
            xlabel('X');
            ylabel('Z');
            zlabel('Y');
            title('Random 100 Points in Point Cloud');
            view([45 30]);
            xlim([-0.5 0.5]);
            ylim([-0.5 0.5]);
            zlim([-0.3 1]);
            pause(0.01); % Pause to display the plot

            % 保存随机选择的100个点的坐标信息
            save('random_100_points.mat', 'randomPointsX', 'randomPointsY', 'randomPointsZ');
        end
    catch e
        % 捕获并处理可能出现的错误
        disp(['Error: ', e.message]);
    finally
        % Stop streaming
        pipe.stop();
    end
end

MartyG-RealSense, 2025-01-04:
I researched your question extensively but could not find an approach to obtaining specific XYZ vertices that was very different from the method that you are already using.  For example:

https://support.intelrealsense.com/hc/en-us/community/posts/4415502202771/comments/4485966351635

lrl-hub, 2025-01-06:
Okay, thank you for your reply.