realsenseai/librealsense #13551: Cmake flags used to build debs

oscarthorn, 2024-11-27:
Hi!

What cmake flags were used when building the packages available in the ppa?

I'm installing from `https://librealsense.intel.com/Debian/apt-repo bionic main` on jetson jetpack 4 / ubuntu 18

Specifically `librealsense2=2.50.0-0~realsense0.6162`. Also wondering if it has changed for newer version? 

I also saw in another issue that debs for jetson were built with cuda enabled? But this seems to have changed at some point since the `librealsense2.so` from `librealsense2=2.50.0-0~realsense0.6162` does not seem to include any symbols related to cuda (checked with `nm -DC /usr/lib/x86_64-linux-gnu/librealsense2.so`?

Thanks!

MartyG-RealSense, 2024-11-27:
Hi @oscarthorn  I do not have information about how the packages are built by my Intel RealSense colleagues.  However, the build configuration page at the link below lists which flags are True by default and so should give an idea of what's automatically enabled in the SDK.

https://dev.intelrealsense.com/docs/build-configuration

Typically, a minimal CMake source code build instruction will just contain the flag **DCMAKE_BUILD_TYPE=release** as the SDK builds in developer 'Debug' mode by default and so does not provide optimizations unless the build type is changed to 'Release'.  Line 46 of the SDK's 'libuvc_installation.sh. pre-made build script for compiling librealsense from source demonstrates this.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

Yes, the Jetson DKMS packages have CUDA support enabled by default, whilst the packages for x86/x64 PCs do not.  That has not changed as far as I am aware.

As you mention newer versions, it is worth highlighting that from librealsense 2.56.1 onwards, Ubuntu 18 and JetPack 4 are both given End of Life (EOL) status in the SDK.

The librealsense SDK will not have CUDA support included in the build when it is compiled from source code unless the flag **-DBUILD_WITH_CUDA=on** is used in the CMake build instruction.  This is why it is necessary to manually include the flag when building for Jetson from source.

oscarthorn, 2024-11-27:
Thanks for the response!

closed 2024-11-27 by oscarthorn