realsenseai/librealsense #13552: Unhandled exception escaping from a worker thread!

nitinpatil5253, 2024-11-27:
| | |
|---|---|
|**librealsense**|2.56.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-11-27:
Hi @nitinpatil5253  Information about this error can be found at https://github.com/IntelRealSense/librealsense/issues/13366

nitinpatil5253, 2024-11-27:
We are using IntelRealSense D457 camera and connected to Windows 11 machine. Latest SDK installed. We are uanble to see the camera in the viewer software. We have connected D457 from a C type port of camera to the USB of laptop. Can you suggest latest setup steps applicable for D457 on windows 11.

We tried running the powershell command from the page given below but, after registry entry, camera is not getting deteced. Below error is coming. Please suggest.

https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide

0 connected RealSense devices were found:

Processing Registry branch HKLM:\SYSTEM\CurrentControlSet\Control\DeviceClasses\{e5323777-f976-4f5b-9b55-b94699c46e44}
There are  total devices listed
0 of them are Intel Realsense

Processing Registry branch HKLM:\SYSTEM\CurrentControlSet\Control\DeviceClasses\{65E8773D-8F56-11D0-A3B9-00A0C9223196}
There are 4 total devices listed
0 of them are Intel Realsense

Task Completed

MartyG-RealSense, 2024-11-27:
The D457 camera model is configured by default to use a GMSL / FAKRA cable connection instead of USB.  In order for it to be detected on USB, you need to move a physical mode-switch in a recess within the camera casing to the 'USB' position using instructions at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

nitinpatil5253, 2024-11-27:
Many thanks for your prompt support. After doing this change, the output is now coming on the viewer.

Can you please confirm if D457 can be used similar to the discontinued product T265, for non-gps navigation?

MartyG-RealSense, 2024-11-27:
You are very welcome.  I'm pleased that I could help.

D457 has a similar IMU motion tracking component to T265 but a key difference is that RealSense 400 Series cameras such as D457 do not have T265's ability to track the camera's own relative position.  So they can be used for SLAM navigation but cannot report their position in the world and how far they have travelled.

nitinpatil5253, 2024-11-28:
Many thanks for your reply. If D457 doesn't have all functionalities of T265, Please suggest alternative for T265 from available cameras, that will help.

MartyG-RealSense, 2024-11-28:
I can only comment on RealSense products, and there is not an alternative that provides built-in device position tracking.

However, there is a software-based possibility that you could use with D457.  You could use ROS and the SLAM navigation tool **rtabmap_ros**, which has a 'localization mode' - please see https://github.com/introlab/rtabmap/discussions/970

Usually with localization though, a map of the environment that the drone / robot is operating in must already have been pre-recorded so that the drone / robot can localize its position via the detail on the map.  So whilst this could work for an indoor drone, it would not be suitable for a flying drone in the sky.

http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping#Localization_mode

nitinpatil5253, 2024-11-28:
Many thanks for the support. It was really useful. Also thanks for sharing the required resources. I appreciate.

nitinpatil5253, 2024-11-28:
Hello @MartyG-RealSense , With respect to my previous question, what was the replacement product for T265 after discontinuation of T265. Please suggest the alternative intel realsense camera which is best alternative to T265 for gps denied navigation?

MartyG-RealSense, 2024-11-28:
There was not an equivalent tracking-specialized product released in the RealSense range after retirement of the T265  In the official End of Life notice in the PDF document at the link below, Intel's recommended replacement strategy was to use a D435i or D455 IMU-equipped camera paired with third-party SLAM software (like the rtabmap_ros tool I mentioned above).

[PCN118463-00.pdf](https://github.com/IntelRealSense/librealsense/files/7100557/PCN118463-00.pdf)

MartyG-RealSense, 2024-12-07:
Hi @nitinpatil5253  Do you require further assistance with this case, please?  Thanks!

nitinpatil5253, 2024-12-08:
Hi @MartyG-RealSense , All the issues have been resolved and the camera is working fine now and we are getting expected output from Realsense library. We are implementing ROS solution for SLAM based navigation using ardupilot, if we get any issue we will get in touch.  Please feel free to close the ticket. Many thanks for your support.

closed 2024-12-08 by nitinpatil5253

MartyG-RealSense, 2024-12-08:
You are very welcome, @nitinpatil5253 - I'm pleased that I could help.  Please do feel free to open a new ticket if you have further issues.  Thanks!