realsenseai/librealsense #13555: Advanced_mode_example

wenmingxiaohuo, 2024-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
D435
Hello expert, thank you for your many enthusiastic answers. I have another question, the source code downloaded from github is not modified, the wrapper/Matlab/advanced_mode_example.m script I understand means to start advanced mode, but it does not implement the function, how can I start the camera's advanced mode through matlab?

MartyG-RealSense, 2024-11-28:
Hi @wenmingxiaohuo  Advanced Mode is already enabled by default on RealSense 400 Series cameras such as D435.  So running the script will not tell you that Advanced Mode is already enabled.  It would just silently proceed to running the json loading code at line 34 of the script.

MartyG-RealSense, 2024-12-07:
Hi @wenmingxiaohuo  Do you require further assistance with this case, please?  Thanks!

wenmingxiaohuo, 2024-12-09:
No, thank you.

MartyG-RealSense, 2024-12-09:
Thanks very much.  As you do not require further assistance, I will close this case.

closed 2024-12-09 by MartyG-RealSense