realsenseai/librealsense #13560: CustomCalibrationTool.exe

Chuanzhaowang, 2024-11-29:
Where can I find the customcalibrationtool.exe file?
When I followed the instructions in intel-realsense-d400-series-custom-calibration-whitepaper.pdf, I did not find the customcalibrationtool.exe file in the customcalibration\bin folder.

Chuanzhaowang, 2024-11-29:
Can you provide complete compiled files (including openCV) or newer documentation there?
When I deploy according to the instructions in "intel-realsense-d400-series-custom-calibration-whitepaper.pdf", there are always some errors!

MartyG-RealSense, 2024-11-29:
Complete compiled files are unable to be provided.

The Dynamic Calibration application is typically updated once per year, though it was not updated in 2024.  

Please provide information about the error messages that you received.  However, the custom calibration API is used so infrequently that there may not be information available about resolving those errors.  Using the pre-made Dynamic Calibration tool instead of the custom API is strongly recommended.