realsenseai/librealsense #13563: [RealSense D435i] How to use emitter on/off mode in practice with Python

zumpchke, 2024-12-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Is the point of emitter on/off mode the ability to get a clean IR image but with accurate depth sensing?

And to do this does the user have to account for account for depth mismatching IR by 1 frame?

MartyG-RealSense, 2024-12-01:
Hi @zumpchke  Having the emitter on (which projects IR light onto surfaces and adds a semi-random invisible dot pattern to the scene) usually improves depth quality, whilst having the emitter off reduces depth quality unless the scene is strongly illuminated.  The camera uses the dots cast onto the surfaces of objects in a scene as a 'texture source' that it can analyze for depth information, as low-texture or no-texture objects (such as doors, walls and tables) will be more difficult for the camera to depth-analyze.

However, for some users' projects it is important for them to have an IR image without a pattern, which is why they capture a frame with the emitter off as well as one with the emitter on.

An IR image will be pixel-perfect aligned, calibrated, and overlapped with the depth map, and also perfectly time-synchronized with it.  It is the **RGB** sensor that is temporally offset from depth, bounded by a period of one frame with the lowest FPS.  

If **wait_for_frames(**) is used in a program script then the SDK tries to find the closest timestamp match between different stream types.

zumpchke, 2024-12-01:
@MartyG-RealSense So is there a way to get simultaneous streams of depth and IR (with **no dots**)

MartyG-RealSense, 2024-12-01:
You might be able to make a difference to the pattern without completely disabling the emitter if you configure the Advanced Mode's **Color Correction Matrix** values.  The matrix is described at the link below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#use-the-left-color-camera

There was also a Visual Preset camera configuration file for doing so called 'Left Imager Color w/o IR Pattern'.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

However, this was designed to be used with the D415 camera model rather than D435i.  If you wanted to try to replicate this feature on D435i, it might be worth setting the **Emitter Always On** option to True so that the D435i's IR emitter behaves like the one on the D415 model.  By default the D415's emitter is Always On, whilst the default state of the D435i's emitter unless Always On is true is to pulse in line with exposure.

Alternatively, if you wanted no dots then you could disable the emitter by setting the value of the **Laser Power** option to '0', but you would lose the advantage of the scene-illuminating IR light that the emitter projects.

closed 2025-01-06 by MartyG-RealSense