realsenseai/librealsense #13564: How is the depth mentioned in D435 calculated?

wenmingxiaohuo, 2024-12-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-12-02:
Hi @wenmingxiaohuo  Pages 18 and 19 of the currenr edition of the data sheet for RealSense 400 Series cameras provides an explanation will illustration images.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

To quote that section of the data sheet:

**********

"The Intel RealSense D400 series depth camera uses stereo vision to calculate depth.  The stereo vision implementation consists of a left imager, right imager, and an optional infrared projector.  The infrared projector projects a non-visible
static IR pattern to improve depth accuracy in scenes with low texture".  

"The left and right imagers capture the scene and send imager data to the depth imaging (vision) processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via the shift between a point on the Left image and the Right image.  The depth pixel values are processed to generate a depth frame.  Subsequent depth frames create a depth video stream."

*********

Intel's **Beginner's Guide to Depth** article may also be a helpful reference.

https://www.intelrealsense.com/beginners-guide-to-depth/

wenmingxiaohuo, 2024-12-03:
Thank you very much for your detailed answer. If I would like to ask what is the sampling frequency of the D435, and also if I want to synchronize the data acquisition with the D435, Lidar L515, how do I set the synchronization information?

MartyG-RealSense, 2024-12-03:
I will reply at the issue https://github.com/IntelRealSense/librealsense/issues/13567 that you created.  Thanks very much!

closed 2025-01-06 by MartyG-RealSense