realsenseai/librealsense #13566: execpetion when connecting camera (L515)

Arulchozhan, 2024-12-03:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
![l515_issue while start](https://github.com/user-attachments/assets/c6726caa-f4e7-4651-bc71-00f225af1121)

```

MartyG-RealSense, 2024-12-03:
Hi @Arulchozhan  The L515 camera model cannot be detected by RealSense SDK version 2.55.1 as L515 was removed in version 2.54.1 because it is a retired camera model.  The L515 should be able to be detected if you install the full **2.53.1** SDK with the 'Intel.RealSense.SDK-WIN10' installer file or download and run the 'Intel.RealSense.Viewer.exe' file.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

MartyG-RealSense, 2024-12-13:
Hi @Arulchozhan  Do you require further assistance with this case, please?  Thanks!

Arulchozhan, 2024-12-16:
@MartyG-RealSense I managed to construct and use the DLL, however, I still can't run the capture example. I'm receiving the following issue. 
[image]
[image]
[image]

MartyG-RealSense, 2024-12-16:
There is advice about this error at https://github.com/IntelRealSense/librealsense/issues/4495

Arulchozhan, 2024-12-17:
@MartyG-RealSense  
[image]
 yes (point 2 ) it works for me thanks for your assistance

MartyG-RealSense, 2024-12-17:
You are very welcome.  It's great to hear that point 2 worked for you!

closed 2025-01-06 by MartyG-RealSense