realsenseai/librealsense #13569: imu校准

wuliu-G, 2024-12-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                   D400 }
| Operating System & Version       Linux (Ubuntu 18                                    | 

当启动校准rs-imu-cabliration.py时，没有移动，但是一直出现warning
[image]

MartyG-RealSense, 2024-12-03:
Hello @wuliu-G  If the camera is on a totally flat surface when using the IMU calibration tool and the values keep changing as though the camera is moving then it is not a problem that can be fixed and it is best to stop using the tool.

It is likely that your IMU is already well calibrated and would not benefit significantly from being calibrated again.  

**********

你好@wuliu-G 如果使用 IMU 校准工具时相机处于完全平坦的表面上，并且数值不断变化，就好像相机在移动一样，那么这不是一个可以修复的问题，最好停止使用该工具。

您的 IMU 可能已经校准得很好，再次校准不会带来太大的好处。

MartyG-RealSense, 2024-12-13:
Do you require further assistance with this case, please?  Thanks!

您需要进一步协助处理此案吗？谢谢！