realsenseai/librealsense #13571: RuntimeError: Frame didn't arrive within 5000 Inside Docker container with Nvidia Jetson ORIN

kais-bedioui, 2024-12-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | v2.50.13 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                        | 
| Platform                                 | NVIDIA Jetson& Docker  |
| SDK Version                            |  { 2.50.13 }                          | 
| Language                            |  python }                          | 
| Segment			|  {Other }                          | 

### Issue Description
Hello,
I have a FRAMOS D455e stereo camera connected to Nvidia Jetson ORIN with Jetpack 5.1 via ethernet port (not PoE).
I successfully created a Docker image to run my Computer Vision application using the camera following the instructions unpacked from _FRAMOS_D400e_Software_Package_Linux64_ARM_latest_ and extracted with the debian package under `~/librealsense2/scripts/Docker_FRAMOS/Dockerfile`.
I can successfully run `rs-enumerate-devices`, `rs-color` and `rs-depth` inside the docker container.
The python example scripts I managed to run and get the expected output from: 

`python3 librealsense2/wrappers/python/examples/python-tutorial-1-depth.py # changed import pyrealsense2 as rs ---> import pyrealsense2.pyrealsense2 as rs`

**BUT** when I apply the same technique to read color frames, as exhibited in the script `opencv_viewer_example.py' I get this error:

`RuntimeError: Frame didn't arrive within 5000`

How can I proceed? Thank you

MartyG-RealSense, 2024-12-04:
Hi @kais-bedioui  Intel does not provide support for the 'e' range of industrial RealSense-compatible cameras as they are produced by FRAMOS and not by Intel.  I recommend contacting FRAMOS support by visiting the link below and clicking on the 'Contact US' button.

https://www.framos.com/en/resources/support

I would emphasize though that **python-tutorial-1-depth.py** is not designed for use with color frames as it is a depth-only script.  It uses an instruction called **get_distance()** to generate a real-world depth value in meters for a particular XY coordinate, but this instruction will only work with the depth stream.

The **align_depth2color.py** example would be a better program to demonstrate accessing color frames.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

kais-bedioui, 2024-12-04:
Hi @MartyG-RealSense 
Thanks for getting back to me. I will get in touch with FRAMOS then.
However I think it is a pyrealsense library-related issue.
I mentioned **python-tutorial-1-depth.py** to emphasize that the Python SDK can successfully read depth frames while other applications that try to read the color frame throw the mentioned error.
Again, I point out that with _realsense-viewer_ I can read both streams and the command **rs-color** can also read the color frames and doesn't throw an error
Could you think of another way of debugging?

MartyG-RealSense, 2024-12-04:
It is difficult to advise, as what works for an Intel D455 and its firmware may not always be the same for a FRAMOS D455e and their own firmware driver.  This is why it is recommended to contact FRAMOS for 'e' camera support.

D455 - at least, the Intel version - supports a mode called 'RGB from left infrared sensor' where you can stream an RGB image from the left infrared sensor instead of an RGB sensor.  This method typically works in cases where the RGB sensor will not stream in Python.  

You can define 'color from left infrared sensor' with a 'cfg' camera configuration line like the one below, which sets the stream format to OpenCV-friendly BGR8 instead of the default infrared Y8 format:

`config.enable_stream(rs.stream.infrared, 640, 480, rs.format.bgr8, 30)`

kais-bedioui, 2024-12-04:
> sensor instead of the RGB sensor. This method typically works in cases

Okay I will also give that a try.
Will keep you posted

kais-bedioui, 2024-12-05:
Hi @MartyG-RealSense 
Just wanted to let you know I solved my issue; accidentally.
I re-ran the docker container with option --runtime nvidia ... in order to test display functionality and that resulted in successfully reading the Color Frames with the python SDK.
I am closing this issue ;)

closed 2024-12-05 by kais-bedioui

MartyG-RealSense, 2024-12-05:
It's great to hear that you achieved a solution!  Thanks very much for the update.  :)

JGiltayOMR, 2024-12-12:
Hi @kais-bedioui ,
How did you manage to create a docker image that can interface with the FRAMOS camera on the Jetson Orin?
I'm struggling to figure it out, got any pointers? 
Thanks in advance

kais-bedioui, 2024-12-12:
@JGiltayOMR 
To build it, I created a Dockerfile based on the Dockerfile you will find under `/usr/src/librealsense2/scripts/Docker_FRAMOS/Dockerfile` after you install Realsense SDK locally

In Line 31 change 'FRAMOS_D400e_Software_Package_Linux64_x64_latest.tar.gz' with 'FRAMOS_D400e_Software_Package_Linux64_ARM_latest'

Then there is the cmake build phase of librealsense2, add ` -DPYTHON_EXECUTABLE=/usr/bin/python3.8 \
    -DBUILD_WITH_CUDA=true \ `
    
 At the second stage of the Docker image also change line 76 to `COPY --from=librealsense_framos-builder /usr/src/librealsense2/build/wrappers/python/pyrealsense2.cpython-38-aarch64-linux-gnu.so*  /usr/lib/python3/dist-packages/pyrealsense2`

This should build you a Docker image via _build_image.sh_ also under `/usr/src/librealsense2/scripts/Docker_FRAMOS/`
You can run it via _run_image.sh_ and add the option `--runtime nvidia`

JGiltayOMR, 2024-12-12:
You're a legend! I'll try that, thanks so much for the quick reply.

JGiltayOMR, 2024-12-12:
Did you have to specify the cuda compiler anywhere? I am trying to set the CUDACXX environment variable to 
usr/local/cuda/bin/nvcc
But it keeps complaining it cant find it.

kais-bedioui, 2024-12-12:
Hey @JGiltayOMR 
Let's first open a new issue, related to your topic. There I will share with you the entire Dockerfile ;)
The issue you are facing is because the BASE_IMAGE is ubuntu:20.04 docker image which doesn't contain CUDA

kais-bedioui, 2024-12-13:
@JGiltayOMR Are you still struggling to build the Docker image or run it?

JGiltayOMR, 2024-12-13:
@kais-bedioui No thanks! Your small comment there made me realise what I had to do! Had to do some other things with the cmakelists.txt for the wrappers as well but eventually I got there! 
Thanks so much for the pointers, that has probably saved me couple of days of frustration!

flxinxout, 2025-03-27:
hello. I have the same issue with an jetson orin nx. I have a d405 and i use pyrealsense2 inside a docker container. My host has jetpack 6 and my docker image is an ubuntu 22.04 with FROM dustynv/pytorch:2.1-r36.2.0

Interestingly, I have another jetson orin nx (exactly the same) where i dont have any errors.

luvwinnie, 2025-07-12:
@flxinxout I'm facing the same problem, do you find the way to solve it?