realsenseai/librealsense #13577: Troubleshooting Intel RealSense D455 Point Cloud Output on JetPack 6.1 with ROS 2

HTLife, 2024-12-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Ubuntu 22.04 (JetPack 6.1)  | 
| Kernel Version (Linux Only)    |  5.15.148-tegra                                        | 
| Platform                                 | NVIDIA Jetson Orin Nano  |
| SDK Version                            |  2.56.3                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

According to the README page of the librealsense library, the latest SDK version (2.56.3 beta) officially supports only JetPack 6.0. However, the JetPack SDK Manager currently provides only versions 5.x.x and 6.1 (but not 6.0).

1. To resolve this, I attempted using JetPack 6.1 to make the camera work.
Initially, I tried using the older camera firmware version 5.13.0.55. Despite modifying the ROS wrapper settings, the ROS 2 wrapper (realsense-ros, Build 4.56.3) was unable to output a point cloud.
2. I then upgraded the camera firmware to 5.16.0.1. After the upgrade, even the realsense-viewer could no longer detect or display the device, though the camera still shows up in the output of lsusb:
```bash
$ lsusb
Bus 002 Device 002: ID 8086:0b5c Intel Corp. Intel(R) RealSense(TM) Depth Camera 455
```

Do you have any suggestions to fix this issue so the D455 camera can output a point cloud correctly using ROS 2 on JetPack 6.1?

MartyG-RealSense, 2024-12-05:
Hi @HTLife  What method did you use to install librealsense on your Jetson, please?

HTLife, 2024-12-08:
Hi @MartyG-RealSense, 
I build it from sources.

MartyG-RealSense, 2024-12-08:
Thanks very much for the information.  

Did you build from source code using the **RSUSB backend = true** method?  Because this installation method bypasses the kernel, it can enable RealSense camera models equipped with an IMU (such as D455) to work correctly on Jetson boards that are using JetPack 6.

Intel's official Jetson installation instruction page recommends using the **libuvc backend** procedure at the link below for an RSUSB source code installation. 

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The procedure provides a pre-made build script called  'libuvc_installation.sh'.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

When using the script with Jetson, it is recommended to edit the CMake build command on line 46 of the script to add the flag **-DBUILD_WITH_CUDA=on**

closed 2025-01-20 by MartyG-RealSense