realsenseai/librealsense #13578: How to enable metadata of IR stream of D457

Deniel-L, 2024-12-05:
we are using D457 GMSL camera on Jetson Agx Orin Jetpack 5.1.2.
The official librealsense implementation skips the metadata of IR and IMU. I'd like to ask how to enable the metadata of IR?

MartyG-RealSense, 2024-12-05:
Hi @Deniel-L  Thanks very much for your question.  I will discuss your issue with my Intel RealSense colleagues.

MartyG-RealSense, 2024-12-06:
Hi @Deniel-L  My colleagues have provided the following advice.

"This is expected.  Hardware embedded Metadata is not supported for IR for D457 GMSL.  Please use the depth metadata which can be shared for depth and IR frameset as every depth is based on left / right IR".

MartyG-RealSense, 2024-12-13:
Hi @Deniel-L  After the information provided in the comment above, do you require further assistance with this case please?  Thanks!

closed 2025-01-06 by MartyG-RealSense