realsenseai/librealsense #13579: what now? Everything was working what is this and how to fix?

p3j6v9, 2024-12-05:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_create_context_ex(api_version:25603, json_settings:{):
    failed to load configuration file (C:\Users\J.Ra\AppData\Roaming\realsense-config.json): [json.exception.parse_error.101] parse error at line 1, column 1: attempting to parse an empty input; check that your input string or stream contains the expected JSON

```

p3j6v9, 2024-12-05:
I have 2 x D455 cameras. tring to get the to work

MartyG-RealSense, 2024-12-05:
Hi @p3j6v9  Please first try using the instructions at https://github.com/IntelRealSense/librealsense/issues/13457#issuecomment-2432900837 to find the **realsense-config.json** file in the Windows `/AppData/Roaming` folder and delete the .json file.

closed 2025-09-07 by MartyG-RealSense