realsenseai/librealsense #13581: Integrating Raspberry Pi 4 with Intel RealSnese D455

amjram03, 2024-12-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

amjram03, 2024-12-06:
Please, I want a step-by-step guide for integrating Raspberry Pi 4.0 with the Intel RealSense D455 camera. Provide the resources used

MartyG-RealSense, 2024-12-07:
Hi @amjram03  When installing the librealsense SDK onto a Raspberry Pi, the best method to try first is the **libuvc backend** installation procedure at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

This uses a pre-made build script called **libuvc_installation.sh**.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

If you require the RealSense Viewer tool and the example programs to be included in the installation then please modify line 46 of the script to use the command below.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

closed 2025-01-20 by MartyG-RealSense