realsenseai/librealsense #13589: D435i and D455f are not recognized by realsense-viewer.

KE-YOSHIOKA-PMT, 2024-12-10:
| | |
|**camera**|REALSENSE D435i D455f|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Ubuntu 22.04|

Hello. An error occurred when I started realsense-viewer. The results of lsusb are as follows:
```
Bus 002 Device 006: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```
/sys/class/video4linux did not exist on my PC.

I'm having trouble because the camera was recognized until the other day. I'd like to know the cause. Neither the D435i nor the D455f could be connected.

```
15:37:29 [Info]  - librealsense version: 2.55.1

15:37:29 [Info] ./src/rs.cpp:3103 - Framebuffer size changed to 1850 x 1016
15:37:29 [Info] ./src/rs.cpp:3103 - Window size changed to 1850 x 1016
15:37:29 [Info] ./src/rs.cpp:3103 - Scale Factor is now 1
15:37:29 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:29 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:29 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:29 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:29 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:29 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:29 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:29 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:29 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:30 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:30 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:30 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:30 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:30 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:30 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:30 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:30 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:30 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:30 [Info] ./src/linux/backend-hid.cpp:1250 - Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
15:37:30 [Info] ./src/linux/backend-v4l2.cpp:646 - Cannot access /sys/class/video4linux
15:37:30 [Info] ./src/context.cpp:128 - Found 0 RealSense devices (0xfe requested & 0xff from device-mask in settings)
15:37:45 [Info] ./src/rs.cpp:3103 - Framebuffer size changed to 925 x 1016
15:37:45 [Info] ./src/rs.cpp:3103 - Window size changed to 925 x 1016

```

KE-YOSHIOKA-PMT, 2024-12-10:
```
Bus 002 Device 007: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0b3a 
  bcdDevice           50.c7
  iManufacturer           1 Intel(R) RealSense(TM) Depth Camera 435i
  iProduct                2 Intel(R) RealSense(TM) Depth Camera 435i
  iSerial                 3 926623051347
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0c7f
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              440mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Intel(R) RealSense(TM) Depth Camera 435i Depth
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              5 Intel(R) RealSense(TM) Depth Camera 435i Depth
      VideoControl Interface Descriptor:
        bLength                14
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength       0x005b
        dwClockFrequency       48.000000MHz
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
        bmControls           0x00200000
      VideoControl Interface Descriptor:
[... 1611 lines cut ...]
    bU2DevExitLat        2047 micro seconds
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  Platform Device Capability:
    bLength                28
    bDescriptorType        16
    bDevCapabilityType      5
    bReserved               0
    PlatformCapabilityUUID    {d8dd60df-4589-4cc7-9cd2-659d9e648a9f}
    CapabilityData[0]    0x00
    CapabilityData[1]    0x00
    CapabilityData[2]    0x03
    CapabilityData[3]    0x06
    CapabilityData[4]    0x44
    CapabilityData[5]    0x02
    CapabilityData[6]    0x20
    CapabilityData[7]    0x00
Device Status:     0x0001
  Self Powered

```

Ash177048, 2024-12-10:
Can you check if you have installed the SDK libraries properly 
`sudo apt-get install libusb-1.0-0-dev libudev-dev libv4l-dev v4l-utils`

and alternately check permission for `/dev` and `/sys/class/video4linux`
`ls -l /dev/video*`

KE-YOSHIOKA-PMT, 2024-12-10:
Thanks for the answer.
I have `libv4l-dev:amd64 (1.22.1-2build1)` installed, but I get the error `ls: cannot access '/dev/video*': No such file or directory`.

Ash177048, 2024-12-10:
Try giving the permission using `sudo chmod 666 /dev/video*`

KE-YOSHIOKA-PMT, 2024-12-10:
```
sudo chmod 666 /dev/video*

chmod: cannot access '/dev/video*': No such file or directory
```

Ash177048, 2024-12-10:
the kernel files are missing try using `sudo modprobe uvcvideo sudo modprobe videodev` and then check again

KE-YOSHIOKA-PMT, 2024-12-10:
An error has occurred
```
sudo modprobe uvcvideo
modprobe: ERROR: could not insert 'uvcvideo': Key was rejected by service
```

```
modinfo uvcvideo | grep version
version:        1.1.1-realsense-1.3.28
srcversion:     62F07434F4A5D0D6EBBBD1D
vermagic:       6.8.0-49-generic SMP preempt mod_unload modversions 
```

MartyG-RealSense, 2024-12-10:
Hi @KE-YOSHIOKA-PMT  **Failed to read busnum/devnum** is a rare error that usually occurs if the camera is being provided by the USB port or USB cable with insufficient power for the camera to operate correctly.  

Do you have a different USB port on your computer that you can try the camera in if you have not done so already, please?

KE-YOSHIOKA-PMT, 2024-12-17:
Sorry for the late reply. Thank you for your answer. I tried the cable that came with the D445f and other USB ports, but it didn't work. I also tried it with Windows 11, but it didn't work.

MartyG-RealSense, 2024-12-17:
The error **Cannot access /sys/class/video4linux** can occur if there is no /`sys/class/video4linux` because there was a problem when trying to insert the uvcvideo kernel-module using modprobe.

Building librealsense from source code with the **libuvc backend** installation procedure at the link below can help to resolve issues related to kernel problems, because a libuvc build of librealsense bypasses the kernel.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvcbackend method uses a pre-made script called 'libuvc_installation.sh' to build librealsense from source code.  If you attempt to use this method and you need to use realsense-viewer then please edit line 46 of the script to change it to the line below:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

********

Are the D435i or D455f detected if you visit the link below on your Windows computer, go to the 'Assets' file list at the bottom of the page and then download and run the file **Intel.RealSense.Viewer.exe**

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

KE-YOSHIOKA-PMT, 2024-12-18:
I reinstalled UbuntuOS and RealSenseSDK and the problem was solved. Thank you for your help.

closed 2024-12-18 by KE-YOSHIOKA-PMT

MartyG-RealSense, 2024-12-18:
You are very welcome, @KE-YOSHIOKA-PMT - it's great to hear that you achieved a solution.  Thanks very much for the update!