realsenseai/librealsense #13594: Turn on both the color channel and the depth channel, an error is thrown-----rs2::backend_error

kummerrrrrr, 2024-12-11:
pipeline pip;
config cfg;
//Only one of them can be enabled
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 15);
pip.start(cfg);
I use a USB 3.0 extension cable, is it possible that the bandwidth is insufficient? I was able to open the D435i camera using the realsense-viewer command.
By the way, I used the camera for YOLOv8 inference, and the input size is (640,640), does this have an effect on the opening of the color stream and the depth stream?
THANKS!!!

MartyG-RealSense, 2024-12-11:
Hi @kummerrrrrr  Are both streams able to display simultaneously if you change the FPS speed for both streams to the minimum supported speed of '6'

Are you running your C++ script and YOLOv8 at the same time?  Only one program at a time can access a particular stream.  For example, if you launched YOLO first and then your C++ script secondly afterwards, the script would not be able to use a stream that YOLO was already using.

MartyG-RealSense, 2024-12-11:
Input / output errors do indeed indicate a possible problem with the USB system (either the USB port or the USB cable).

Are you using the official 1 meter long USB cable supplied with the camera or another USB cable that you supplied yourself, please?

kummerrrrrr, 2024-12-11:
> Input / output errors do indeed indicate a possible problem with the USB system (either the USB port or the USB cable).
> 
> Are you using the official 1 meter long USB cable supplied with the camera or another USB cable that you supplied yourself, please?

I used a USB 3.0 extension cable, 3 meters long, connected to the official 1 meter long USB cable (due to venue reasons, my computer cannot be connected to the official cable)

MartyG-RealSense, 2024-12-11:
When using USB cables longer than 2 meters it is imporant that the cable is high quality, otherwise performance can be unstable or the camera may disconnect.  Cables should ideally also be 'USB 3.1 Gen 1' rather than 3.0, as RealSense 400 Series cameras only minimally supported 3.0 even when they were first released 7 years ago.

kummerrrrrr, 2024-12-19:
> When using USB cables longer than 2 meters it is important that the cable is high quality, otherwise performance can be unstable or the camera may disconnect. Cables should ideally also be 'USB 3.1 Gen 1' rather than 3.0, as RealSense 400 Series cameras only minimally supported 3.0 even when they were first released 7 years ago.

hi 
It's me again, I changed the USB3.2 cable or even USB4, but my program still doesn't work, and I still can't turn on both streams at the same time.
This time I'll just use the simplest cpp example，and error prompts：
Ubuntu20.04，SDK version 2.55.1, firmware version 5.16

#include <librealsense2/rs.hpp>
#include <librealsense2/rsutil.h>
#include <librealsense2/hpp/rs_options.hpp>
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;
using namespace rs2;

int main()
{
    pipeline pip;
    config cfg;
    //Can't turn on two streams at the same time, even if they're running at minimum performance
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

    pip.start(cfg);

    while (true)
    {
        frameset frames = pip.wait_for_frames();
        frame color_frame = frames.get_color_frame();
        depth_frame depth = frames.get_depth_frame();

        if (!color_frame)
            continue;

        Mat color_image(Size(640, 480), CV_8UC3, (void *)color_frame.get_data(), Mat::AUTO_STEP);

        imshow("Realsense view", color_image);

        if (waitKey(1) == 27)
        {
            break;
        }
    }
    destroyAllWindows();
    return 0;
}

Thread 1 "my_program" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50

MartyG-RealSense, 2024-12-19:
Are you able to stream depth and color simultaneously in the realsense-viewer tool, as you mntioned earlier that you were able to open the camera in it.

It looks as though you are using C# code.  Is the RealSense SDK's official 'Capture' C# example program at the link below that uses both depth and color able to run successfully?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/tutorial/capture

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/tutorial/capture/Window.xaml.cs#L78-L79