realsenseai/librealsense #13608: File opening - error message

Kravarik-VUJE, 2024-12-16:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2024-12-16:
Hi @Kravarik-VUJE  There was an issue in librealsense 2.54.2 with this error occurring when playing back bag files that were recorded in versions older than 2.54.2.  The issue was fixed in version 2.55.1 onwards.

Kravarik-VUJE, 2024-12-16:
> UPOZORNENIE: Tento email pochádza z externej adresy. S vloženými odkazmi a prílohami pracujte opatrne. Ak sa Vám zdajú podozrivé, nahláste to na  ***@***.******@***.***>
>
> Od adresy ***@***.******@***.***> nedostávate často e-maily. Zistite, prečo je to dôležité<https://aka.ms/LearnAboutSenderIdentification>

Hi,
Thank you for your quick answer and advice.
Kamil

Táto správa obsahuje len stanoviská alebo informácie odosielateľa, ktoré v žiadnom prípade nezakladajú právny vzťah VUJE, a. s. Stanoviská a informácie uvedené v tejto správe sa nemusia zhodovať s oficiálnymi stanoviskami alebo informáciami VUJE, a. s.
Osobné údaje spracúvame a ochraňujeme v súlade s Nariadením Európskeho parlamentu a Rady EÚ 2016/679 o ochrane fyzických osôb pri spracúvaní osobných údajov a o voľnom pohybe takýchto údajov (GDPR) a zákonom č. 18/2018 Z. z. o ochrane osobných údajov a o zmene a doplnení niektorých zákonov v znení neskorších predpisov vrátane jeho vykonávacích predpisov a v súlade s ostatnými všeobecne záväznými právnymi predpismi. Viac o ochrane a spracúvaní osobných údajov je uverejnené na stránke https://www.vuje.sk/ v sekcii „Užitočné linky / GDPR“.

This message contains only opinions or information of the sender, which can not in any case establish legal engagement of VUJE, Inc. Opinions and information included in this message need not be identical with the official opinions or information of VUJE, Inc.
Personal data are processed and protected in accordance with the General Data Protection Regulation (Regulation (EU) 2016/679) and Act no. 18/2018 on Personal Data Protection and amending and supplementing certain Acts as amended and in accordance with generally binding legal regulations. Detailed information about the protection and processing of personal data is published on the website https://www.vuje.sk/ in the section Useful Links / GDPR.

closed 2024-12-16 by Kravarik-VUJE