realsenseai/librealsense #13611: D435i can't be found, ubuntu22.04 jammy(core 5.10), ROS2 humble

Frank198002, 2024-12-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (5.16) | 
| Operating System & Version |   {Linux (Ubuntu 22.04)   | 
| Kernel Version (Linux Only)    |  (5.10.110-rockchip-rk3588)                                         | 
| Platform                                 | OrangePi 5B  |
| SDK Version                            |  {2.56.3 }                          | 
| Language                            |  {C/python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
When install SDK 2.0, I select "Option 3: Build from source".  Follow the instructions under [Linux Installation] and Install librealsense2, on step 3: Build and apply patched kernel modules for:

[image]

MartyG-RealSense, 2024-12-17:
Hi @Frank198002  It is recommended that on Rockchip (RK) devices, the librealsense SDK is built from source code with the **libuvc backend** installation method.  A libuvc build of the SDK bypasses the kernel and so can avoid problems that may occur on a kernel-patched installation of librealsense.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend installation method provides a pre-made build script called 'libuvc_installation.sh'.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

If you need your librealsense installation to include the realsense-viewer and other tools and examples then please change line 46 of the script from this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

to this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

After installing librealsense with the libuvc backend method, please build the RealSense ROS2 wrapper again.

Frank198002, 2024-12-19:
Hi @MartyG-RealSense

MartyG-RealSense, 2024-12-19:
Does it make a differene if you launch rs-enumerate-devices in sudo admin permissions mode with **sudo rs-enumerate-devices**

Ideally, the ROS wrapper should be the only program installed in the **/ros2_ws** catkin workspace directory and librealsense should be installed in another directory.

Frank198002, 2024-12-20:
Here is the new system, realsense is the first application.
librealsense dir: ~/librealsense_build/librealsense-master/
ROS wrapper dir: ~/ros2_ws/
Now realsense-ros is the unique dir in ros2_ws/src as following:
orangepi@orangepi5b:~/ros2_ws/src$ ls
realsense-ros

Thank you for your suggestion! We can see info with "sudo". But realsense-ros still can't find device.
orangepi@orangepi5b:~/ros2_ws$ sudo rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	233522071798
    Firmware Version              : 	5.16.0.1
    Recommended Firmware Version  : 	5.16.0.1
    Physical Port                 : 	7-1-9
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	D400
    Asic Serial Number            : 	226523132791
    Firmware Update Id            : 	226523132791
    Dfu Device Path               : 	

Stream Profiles supported by Stereo Module
 Supported modes:
    STREAM      RESOLUTION     FORMAT      FPS
    Infrared 1  1280x720       Y8          @ 6 Hz
    Infrared 1   848x480       Y8          @ 10/8/6 Hz
    Infrared 1   640x480       Y8          @ 30/15/6 Hz
    Infrared 1   640x360       Y8          @ 30 Hz
    Infrared 1   480x270       Y8          @ 60/30/15/6 Hz
    Infrared 2  1280x720       Y8          @ 6 Hz
    Infrared 2   848x480       Y8          @ 10/8/6 Hz
    Infrared 2   640x480       Y8          @ 30/15/6 Hz
    Infrared 2   640x360       Y8          @ 30 Hz
    Infrared 2   480x270       Y8          @ 60/30/15/6 Hz
    Depth       1280x720       Z16         @ 6 Hz
    Depth        848x480       Z16         @ 10/8/6 Hz
    Depth        640x480       Z16         @ 30/15/6 Hz
    Depth        640x360       Z16         @ 30 Hz
    Depth        480x270       Z16         @ 60/30/15/6 Hz
    Depth        256x144       Z16         @ 90 Hz

Stream Profiles supported by RGB Camera
 Supported modes:
    STREAM      RESOLUTION     FORMAT      FPS
    Color       1920x1080      RGB8        @ 8 Hz
    Color           |          Y8          @ 8 Hz
    Color           |          BGRA8       @ 8 Hz
    Color           |          RGBA8       @ 8 Hz
    Color           |          BGR8        @ 8 Hz
    Color           |          YUYV        @ 8 Hz
    Color       1280x720       RGB8        @ 15/10/6 Hz
    Color           |          Y8          @ 15/10/6 Hz
    Color           |          BGRA8       @ 15/10/6 Hz
    Color           |          RGBA8       @ 15/10/6 Hz
    Color           |          BGR8        @ 15/10/6 Hz
    Color           |          YUYV        @ 15/10/6 Hz
    Color        640x480       RGB8        @ 30/15/6 Hz
    Color           |          Y8          @ 30/15/6 Hz
    Color           |          BGRA8       @ 30/15/6 Hz
    Color           |          RGBA8       @ 30/15/6 Hz
    Color           |          BGR8        @ 30/15/6 Hz
    Color           |          YUYV        @ 30/15/6 Hz
    Color        424x240       RGB8        @ 60/30/15/6 Hz
    Color           |          Y8          @ 60/30/15/6 Hz
    Color           |          BGRA8       @ 60/30/15/6 Hz
    Color           |          RGBA8       @ 60/30/15/6 Hz
    Color           |          BGR8        @ 60/30/15/6 Hz
    Color           |          YUYV        @ 60/30/15/6 Hz

Stream Profiles supported by Motion Module
 Supported modes:
    STREAM      FORMAT         FPS
    Accel       MOTION_XYZ32F  @ 200/100 Hz
    Gyro        MOTION_XYZ32F  @ 400/200 Hz

orangepi@orangepi5b:~/ros2_ws$ source install/setup.bash 
orangepi@orangepi5b:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true
[INFO] [launch]: All log files can be found below /home/orangepi/.ros/log/2024-12-20-10-08-27-774495-orangepi5b-17605
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [17615]
[realsense2_camera_node-1] [INFO] [1734660507.975677600] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1734660507.975803890] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1734660507.975821098] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [WARN] [1734660508.001104701] [camera.camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1734660514.025589473] [camera.camera]: No RealSense devices were found!

MartyG-RealSense, 2024-12-20:
Please try enabling only Motion Module on its own in realsense-viewer to see whether the IMU data appears.  There is a known issue with realsense-viewer for some RealSense users where if you enable three streams and one of them is the Motion Module (such as Depth, RGB and Motion Module), one of the streams will not provide data.  For users of realsense-viewer who experience this problem, there is no fix for this except to only enable Motion Module on its own, or only enable two streams (Depth + Motion Module or RGB + Motion Module).

In the ROS wrapper, the IMU topics are disabled by default unless you add the parameters below to your ros2 launch instruction to publish them.

`enable_accel:=true enable_gyro:=true unite_Imu_method:=2`

Frank198002, 2024-12-22:
I only open IMU stream in viewer as you said, but still can't see IMU image. I remmber when I first install camera,which firmware is 5.13, realsense-viewer can show RGB, depth and IMU at the same time.

a. Here is the error log when i only open IMU in realsense-viewer.
[image]
At first, there is no stream, and wait for a while, it will report "control-transfer returned error "

b. then I close realsense-viewer, resatrt system, and us ROS2 command to run package as following:
```
ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_accel:=true enable_gyro:=true unite_imu_method:=2

********
[realsense2_camera_node-1]  22/12 16:34:06,441 WARNING [547440290016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
```
```
then I check topics as followings:
orangepi@orangepi5b:~/ros2_ws$ ros2 topic list
/camera/camera/accel/imu_info
/camera/camera/accel/metadata
/camera/camera/accel/sample
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_accel
/camera/camera/extrinsics/depth_to_color
/camera/camera/extrinsics/depth_to_gyro
/camera/camera/gyro/imu_info
/camera/camera/gyro/metadata
/camera/camera/gyro/sample
/camera/camera/imu
/parameter_events
/rosout
/tf_static
orangepi@orangepi5b:~/ros2_ws$ ros2 topic echo /camera/camera/color/image_raw

orangepi@orangepi5b:~/ros2_ws$ ^C
orangepi@orangepi5b:~/ros2_ws$ ros2 topic echo /camera/camera/color/metadata
The message type 'realsense2_camera_msgs/msg/Metadata' is invalid
orangepi@orangepi5b:~/ros2_ws$ ros2 topic echo /camera/camera/color/camerainfo
WARNING: topic [/camera/camera/color/camerainfo] does not appear to be published yet
Could not determine the type for the passed topic
```
I can't find the topic output.  But it is surprise that I close ros2 node and open realsense-view, I can see RGB and depth video.  My request is receiving RGB and IMU ros stream from this camera, if depth ok,  it will be best!

MartyG-RealSense, 2024-12-22:
At https://github.com/IntelRealSense/librealsense/issues/13611#issuecomment-2553025976 earlier in this discussion you said that you could use realSense-viewer to see RGB, Stereo and Motion Module.  But when you plugged the camera into USB 3.0 on your Orange Pi, that's when your problems occurred.

Looking at information about Orange Pi 5b, it has 1 USB 3.0 port and 1 USB 3.1 Gen 1 port (plus two USB 2 ports).  It is highly recommendable to ensure that the camera is in the USB 3.1 port.  This is because RealSense 400 Series cameras only minimally supported USB 3.0 even when they were first launched in December 2017.  They should be used with USB 3.1 Gen 1.

Frank198002, 2024-12-22:
Yes, before install D435i, i read the relative issue about USB3.0;  OrangePi5B has 2 USB2.0 and 1 USB3.0. I purchase one USB3.0 wire to connect.
When I use release-viewer , I will connect mouse and keyboard to One usb2.0 ,and D435i to USB3.0(Viewer display "usb3.2"), it will  shows "control-transfer returned error " about once in ten messages.  
But when I use ros2 command, I remote ssh OP5B and USB3.0 for D435i， both of USB2.0 ports are empty.  I will receive a lot of  "control-transfer returned error ".
```
$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_accel:=true enable_gyro:=true unite_imu_method:=2
[INFO] [launch]: All log files can be found below /home/orangepi/.ros/log/2024-12-22-23-32-12-001443-orangepi5b-32320
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [32321]
[realsense2_camera_node-1] [INFO] [1734881532.212810785] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1734881532.213191988] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1734881532.213285028] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1734881532.749153374] [camera.camera]: Device with serial number 233522071798 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1734881532.749326622] [camera.camera]: Device with physical ID 6-1-11 was found.
[realsense2_camera_node-1] [INFO] [1734881532.749384954] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1734881532.751045389] [camera.camera]: Device with port number 6-1 was found.
[realsense2_camera_node-1] [INFO] [1734881532.751130846] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1734881532.751182470] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1734881539.474003869] [camera.camera]: Device with serial number 233522071798 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1734881539.474169241] [camera.camera]: Device with physical ID 6-1-12 was found.
[realsense2_camera_node-1] [INFO] [1734881539.474213283] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1734881539.475553472] [camera.camera]: Device with port number 6-1 was found.
[realsense2_camera_node-1] [INFO] [1734881539.475606846] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1734881539.475850676] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1734881539.477987104] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1734881539.478113977] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1734881539.478205268] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1734881539.478293642] [camera.camera]: Device Serial No: 233522071798
[realsense2_camera_node-1] [INFO] [1734881539.478380557] [camera.camera]: Device physical port: 6-1-12
[realsense2_camera_node-1] [INFO] [1734881539.478467181] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1734881539.478563429] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1734881539.478834676] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1734881542.383779261] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1734881542.386540722] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1734881542.522259180] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1734881542.523162750] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1734881542.523242666] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1734881542.526616910] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1734881542.526762449] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1734881542.526797449] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1734881542.604336562] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1734881542.703155082] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  22/12 23:32:22,703 WARNING [547004344544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
```
Could you kindly give me more suggestions?

Frank198002, 2024-12-23:
Hi MartyG, I find the cause. D435i Firmware should be 5.13.0.55, it runs OK not only in realsense-view but also in ROS2 topic.
[image]
In ROS2, I can see the content in  topic ../../../sample 
The next problem is what is mean for "Frame drop rate in one second". It shows 78%, but I still receive IMU info.
And could you tell me the proper SDK version for FW-5.13.0.55, I didn't know whether SDK 2.55.1 can match FW-5.13.0.55

MartyG-RealSense, 2024-12-23:
I'm pleased to hear that firmware 5.13.0.55 worked for you!

The 'Frame Drops Per Second' graph in realsense-viewer can be disregarded as it does not provide useful information about analysing performance.  Clicking the 'i' button on the top of the stream panels to overlay information over the panel provides more reliable information, such as the current frames per second (FPS) speed of a stream.

The librealsense versions that match to firmware 5.13.0.55 are 2.50.0 and 2.51.1.  However, as you found yourself, 5.13.0.55 can also be successfully used with librealsense 2.55.1.

closed 2025-01-20 by MartyG-RealSense