realsenseai/librealsense #13617: How to implement taking a photo (to obtain depth images) in Android?

1programmer2dfg3, 2024-12-19:
I need to capture images from a depth camera and be able to set the image resolution. How can I achieve this?

MartyG-RealSense, 2024-12-19:
Hi @1programmer2dfg3  The RealSense Android wrapper's **Capture** example detailed at https://github.com/IntelRealSense/librealsense/issues/8551#issuecomment-796714892 may meet your needs.

In the Capture example, you can edit the **config.enableStream** instructions at lines 163 and 164 of the example to set your own preferred resolution for the program to use.

[image]

1programmer2dfg3, 2024-12-20:
Thanks, how do I get the HD image of my current camera ？the bitmap

MartyG-RealSense, 2024-12-20:
https://github.com/IntelRealSense/librealsense/issues/4455 and https://github.com/IntelRealSense/librealsense/issues/5446 may be helpful references for writing a bitmap on Android.

1programmer2dfg3, 2024-12-20:
Look at the comments there is a Mat this class I test here in the Android SDK is not, where does this come from, can you help me to write a complete depth image bitmap example? The previous comments were a bit vague, thank you

MartyG-RealSense, 2024-12-20:
If a 'native' Android application is written then functions in the librealsense SDK's C++ API can be accessed from the Android app.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/readme.md

There is a further Android bitmap creation reference at https://github.com/IntelRealSense/librealsense/issues/6089 and the RealSense user in that case shares their solution at https://github.com/IntelRealSense/librealsense/issues/6089#issuecomment-602651364

There is also an Android project that converts RealSense frames to OpenCV mat and then converts to a bitmap at https://github.com/mcanyucel/realsense-android-combined:

There is a discussion too about bitmap creation here:

https://stackoverflow.com/questions/71381082/converting-16-bit-depth-frame-from-intel-realsense-d455-to-opencv-mat-in-android

There are not many eamples of creating bitmaps with the RealSense Android wrapper though, and I am not able to provide advice about programming a complete RealSense Android wrapper program unfortunately as my programming knowledge of that wrapper is limited.  I hope that the additional references in this comment will help you to make progress in your project.

closed 2025-01-24 by MartyG-RealSense