realsenseai/librealsense #13633: This problem occurs on different devices: the camera image is incomplete.

ZZ-0909, 2024-12-27:
This problem did not exist in my previous device, but recently the Ubuntu system had a problem and I reinstalled the system. After installing realsense again, the camera image (including colour and depth) was incomplete in the realsense-viewer, and the imu image was frozen. What caused this and how can I solve it? In addition, I changed two terminals and the same problem occurred. Thank you for your guidance.
[image]

MartyG-RealSense, 2024-12-27:
Hi @ZZ-0909  To generate a 3D 'pointcloud' image like the one above, at least one of the streams that is enabled needs to be depth (the Stereo Module), otherwise it will not work.  IMU data is also not used in a 3D pointclould.

To display depth, RGB and IMU data correctly, please click the '2D' option in the top corner of the RealSense Viewer window.

MartyG-RealSense, 2025-01-20:
`Case closed due to no further comments received.`

closed 2025-01-20 by MartyG-RealSense