realsenseai/librealsense #13637: How to sync images coming from multiple D435 ?

Akumar201, 2024-12-28:
Hello, I am using Intel RealSense D435 cameras as wrist cameras for bimanual robotic arms, capturing video at 60fps. Currently, I am storing the data from both cameras, but when I replay the videos, they are not synchronized. Is there an inbuilt function in RealSense that ensures the feeds from multiple cameras remain synchronized during recording? I am using suggested cable from https://newnex.com/realsense-3d-camera-connectivity.php

MartyG-RealSense, 2024-12-28:
Hi @Akumar201  You can synchronize depth data of multiple cameras using the RealSense SDK's **hardware sync** feature, though it involves purchasing individual electronic components and assembling them into a 'sync cable' to connect the cameras together via concealed sync pins under a small removable cap on top of the camera casing.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

Akumar201, 2024-12-28:
@MartyG-RealSense , thank you for your response, I have a follow up question is there any way to synchronize images data from Intel RealSense D405 Webcam ? I want to take video of the screen from different camera and sync them together as currently the issue with this is it's not syncing.

MartyG-RealSense, 2024-12-28:
D405 is not equipped with a set of hardware sync pins the other 400 Series models have that enable sync between multiple cameras.

It is possible to sync data from different cameras using **camera metadata** though.  Information about this can be found at https://github.com/IntelRealSense/librealsense/issues/2186

closed 2025-01-20 by MartyG-RealSense