realsenseai/librealsense #13638: Cannot connect camera L515

jayasurya00098, 2024-12-28:
I Recently bought a L515 devices, when I connect to other camera software it shows the devices. On RealSense Viewer, I show the following warnings and doesn't show in Add Sources.

[image]

MartyG-RealSense, 2024-12-28:
Hi @jayasurya00098  The L515 camera model is no longer supported in recent RealSense SDK and RealSense Viewer versions as it is a retired camera model.  It should be detected if you install version **2.53.1** of the SDK or Viewer, the final version to support L515.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

You can install this SDK with an automated installer program called **Intel.RealSense.SDK-WIN10** (which also works with Windows 11) or just the RealSense Viewer by downloading and running the file **Intel.RealSense.Viewer.exe**

[image]

MartyG-RealSense, 2025-01-06:
Hi @jayasurya00098  Do you require further assistance with this case, please?  Thanks!

jayasurya00098, 2025-01-06:
Hi @MartyG-RealSense Thank you, It worked.

closed 2025-01-06 by jayasurya00098