realsenseai/librealsense #13639: How does an Intel D455 camera take a RGB pictures with depth information

junmoxiao11, 2024-12-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Win 11  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |                            | 
| Language                            |  python }                          | 
| Segment			|  {others }                          | 

### Issue Description
How does an Intel D455 camera take a RGB pictures with depth information. Take pictures instead of videos.

MartyG-RealSense, 2024-12-29:
Hi @junmoxiao11  If you wish to export a depth frame to an image file then https://github.com/IntelRealSense/librealsense/issues/2877 discusses saving the frame to a .PNG image file with Python code.

Most of the depth information is lost when saving to .PNG though.  You can instead preserve the depth information by saving to .RAW format instead, as demonstrated in the Python code at https://github.com/IntelRealSense/realsense-ros/issues/1312

junmoxiao11, 2024-12-29:
In the example you gave me, I didn't find the answer I was looking for. What I need is to take a picture like .PNG foemat, which can be clicked directly to open and crop the picture. And the picture must contain depth information.  After I crop the image, the pixels still retain the corresponding depth value.

MartyG-RealSense, 2024-12-29:
The RealSense SDK does not have built-in functions for cropping an image horizontally and vertically.  You would likely have to use OpenCV code to implement that.

junmoxiao11, 2024-12-30:
Do you have any python code to crop these pictures?

MartyG-RealSense, 2024-12-30:
https://github.com/IntelRealSense/librealsense/issues/6267#issuecomment-618261685 suggests a Python approach to XY image cropping using **Open3D**

The link below has an OpenCV approach to cropping.

https://stackoverflow.com/questions/68576101/i-want-to-crop-a-depth-frame-and-get-a-cropped-depth-frame-so-that-i-can-apply-s

junmoxiao11, 2024-12-30:
Are the left and right cameras of the Realsense D455 camera based on visible or infrared imaging?

MartyG-RealSense, 2024-12-30:
They are infrared sensors.  However, on the D455 model the left infrared sensor can also provide an RGB image.

closed 2025-01-20 by MartyG-RealSense