realsenseai/librealsense #13640: MacOS M2 CMake error "Project 'realsense2' requires C++14 or higher"

Aterox-os, 2024-12-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | | 
| Operating System & Version | MacOS Sequoia | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | MacBook Air M2  |
| SDK Version                            |                          | 
| Language                            |  Python, C++                        | 
| Segment			|                         | 

Hello,
I am trying to build Librealsense and Pyrealsense2 on my M2 MacBook Air. The home-brew installation also didn't work since every time I plug the camera in the Realsense viewer crashes. There for I tried building from the source. Unfortunately I am getting bombarded with error. I am following this very nice [Guide](https://lightbuzz.com/realsense-macos/). But currently I am getting the following error:
```
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Failed
CMake Error at CMake/lrs_macros.cmake:17 (message):
  Project 'realsense2' requires C++14 or higher
Call Stack (most recent call first):
  CMakeLists.txt:51 (config_cxx_flags)
```
This error occurs when I try to make the project. Before that I ran the following command:
```
cmake .. -DCMAKE_OSX_ARCHITECTURES="arm64;x86_64" \
-DCMAKE_THREAD_LIBS_INIT="-lpthread" \
-DCMAKE_BUILD_TYPE=RELEASE \
-DBUILD_SHARED_LIBS=ON \
-DBUILD_EXAMPLES=false \
-DBUILD_WITH_OPENMP=false \
-DHWM_OVER_XU=false \
-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl \
-G Xcode
```
The cmake command ran without a problem. I also tried the solution suggegested by na-trium-144 (#11978). But without success. 
If I run `clang --version` the return is:
```
Apple clang version 16.0.0 (clang-1600.0.26.6)
Target: arm64-apple-darwin24.1.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
Has anyone also faced this issue an could fix it?

MartyG-RealSense, 2024-12-30:
Hi @Aterox-os  Which version of the librealsense SDK are you building from source code, please?  From SDK 'Beta 2.56.1' onwards, a fix for compilation on Apple 'M' computers and an upgrade to use C++14 were added, as described at https://github.com/IntelRealSense/librealsense/pull/12972

[image]

Aterox-os, 2024-12-30:
Thank you. With the new version the build process worked just fine. Unfortunately I am encountering the same error as if I used homebrew. Every time I start the RealSense viewer this error appears:
```
GLFW Driver Error: Cocoa: Regular windows do not have icons on macOS
2024-12-30 16:21:17.730 realsense-viewer[49213:1358291] +[IMKClient subclass]: chose IMKClient_Modern
2024-12-30 16:21:17.730 realsense-viewer[49213:1358291] +[IMKInputSession subclass]: chose IMKInputSession_Modern
```
But the viewer starts just fine. But if I plug in my camera the viewer crashed with the following error:
`zsh: segmentation fault  realsense-viewer`

MartyG-RealSense, 2024-12-30:
Does realsense-viewer work correctly if it is launched in sudo admin permissions mode with the command **sudo realsense-viewer**

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/13477 who experienced the `Cocoa: Regular windows do not have icons on macOS` error on MacOS 15.1 said that the camera had worked in MacOS 15.0.  Others had similar problems to you, where realsense-viewer opens but the streams cannot be accessed.  

Some users decided to switch to using Windows (for example, in a virtual environment on Mac instead of using the camera in MacOS).  Another user found that they could eventually access the streams if they performed 'run with root' multiple times.

Aterox-os, 2024-12-30:
I tried running it with sudo but the error is still appearing but with one missing:
```
GLFW Driver Error: Cocoa: Regular windows do not have icons on macOS
2024-12-30 17:29:22.553 realsense-viewer[49270:1364330] +[IMKClient subclass]: chose IMKClient_Modern
zsh: segmentation fault  sudo realsense-viewer
```
But as you can see I still get the segmentation fault when connecting the camera. 
I tried a windows VM with parallels and it worked but I really want to get it working under MacOS.

MartyG-RealSense, 2024-12-30:
My Intel RealSense colleague who worked on the recent Mac fixes in 2.56.1 commented at https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-2326494668 about how segmentation faults are still occurring after the fixes and that it may be due to a change in USB policy by Apple.  If that is the case then it may be difficult to resolve the issue from Intel's end.

closed 2024-12-30 by Aterox-os

alifeinbinary, 2024-12-31:
Hi @MartyG-RealSense 

I installed 2.56.3 from homebrew and ran ./realsense-viewer with sudo privileges and the camera connected to a USB C port on my 2019 Intel MacPro running MacOS 15.2. In the terminal immediately after the password entry I have the following: 

```
GLFW Driver Error: Cocoa: Regular windows do not have icons on macOS
2024-12-31 00:18:52.237 realsense-viewer[68798:1262023] +[IMKClient subclass]: chose IMKClient_Modern
```

The app took a bit of time to load but with sudo privileges at least it didn't crash. Upon the app loading a dialog modal appeared with the title "Oops, something went wrong!" and a Realsense error calling: 

```
Invalid Value in rs2_get_video_stream_intrinsics(profile:[ C RGB8 0 1280x720
@ 30 ], intr:(out)):
set_xu(ctrl=1) failed! Last Error: Operation timed out 
```

When I clicked to activate the depth camera, I saw the following error in a dialog modal: 

```
UNKNOWN in rs2_open_multiple(sensor:0x600001c04060,
profiles:0x60000033eb50, count:1):
failed to set power state
```

and the following the realsense-viewer console:
```
00:30:45 [Error] /tmp/librealsense-20241215-5131-1am9fr/librealsense-2.56.3/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
00:30:45 [Error] /tmp/librealsense-20241215-5131-1am9fr/librealsense-2.56.3/src/uvc-sensor.cpp:428 - acquire_power failed: failed to set power state
00:30:45 [Error] /tmp/librealsense-20241215-5131-1am9fr/librealsense-2.56.3/src/rs.cpp:256 - [rs2_open_multiple( sensor:0x600001c04060, profiles:0x60000033eb50, count:1 ) UNKNOWN] failed to set power state
```

Hope that helps. Would love to see this get resolved in 2025 if at all possible so I can continue development on a side project that uses this amazing camera. 

Happy new year :)

MartyG-RealSense, 2024-12-31:
Thanks so much @alifeinbinary for sharing your detailed feedback, and happy New Year to you too!

alifeinbinary, 2024-12-31:
@MartyG-RealSense you're not going to believe this, it's a Christmas miracle but I got it working on MacOS 15.2!

Here's how I did it:

1. clone this repo
2. `git checkout 05d6eb7ec8efeed4d69b4025d9ff149e7132acd5` <--2.56.3
3. `mkdir build && cd build`
4. `brew install libusb` <-- should be the latest version 1.0.27
5. `brew info libusb` <-- verify version 1.0.27 is installed
6. `sudo xcode-select --reset`
7. `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false DHWM_OVER_XU=false DFORCE_RSUSB_BACKEND=true`
8. `make -j32` <-- I have a 16 core workstation, modify this to suit your system
9. `install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.27.dylib @rpath/libusb-1.0.27.dylib librealsense2.dylib`
10. `cd Release`
11. `sudo ./realsense-viewer` with the camera connected to the computer

It ran successfully indefinitely until I quit the application and now it won't load again but this is progress nonetheless. I'll try again after a system reboot. Would be great if a M Mac user could test this method.

MartyG-RealSense, 2024-12-31:
@alifeinbinary  It's great news that you achieved progress!  Mac users sharing their knowledge with other users and with the Intel RealSense team is the best way to improve installation success on modern MacOS versions and Apple Silicon 'M' Macs.

Aterox-os, 2024-12-31:
@alifeinbinary I tried you solution but unfortunately I couldn't it working. The error remains even on the first startup the same:
`zsh: segmentation fault  sudo ./realsense-viewer `