realsenseai/librealsense #13641: Enabling gyro/acccel causes issues

RikisuT, 2024-12-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.16.0.1| 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-127-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.55.1.0 }                          | 
| Language                            | ROS1 | 
| Segment			|   | 

### Issue Description
I have the above installed but when running 
`ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true`
i get 
```

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Rikisu-2:35445/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
[... 321 lines cut ...]
 30/12 11:47:06,450 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,508 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,517 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,575 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,583 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,642 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,650 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,708 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,717 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,775 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,784 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,834 ERROR [140372568110848] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 30/12 11:47:06,842 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,901 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:06,959 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:07,018 WARNING [140372349024000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 11:47:07,137 ERROR [140372315453184] (error-handling.cpp:78) Error during polling error handler: get_xu(id=7) failed! Last Error: Success
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
Happens only when I enable accel or gyro
When i run `ros2 launch realsense2_camera rs_launch.py` it shows up fine and I'm able to verify it by looking into rviz .

MartyG-RealSense, 2024-12-30:
Hi @RikisuT  Thanks very much for your question.

Do the **control_transfer returned** warnings still occur if you add `unite_imu_method:=linear_interpolation` to your roslaunch instruction to publish the accel and gyro messages to a single combined **/imu** topic instead of separate accel and gyro topics?

If using unite_imu_method does not work, does it make a difference if you add **initial_reset:=true** to the roslaunch instruction to reset the camera at launch?

RikisuT, 2024-12-30:
Tried both, I still get the same error.

MartyG-RealSense, 2024-12-30:
Thanks very much.  Next, please try the following procedure.

1.  Install support for an **RGBD** launch with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Launch the **rs_rgbd.launch** launch file with IMU enabled using this command:

`roslaunch realsense2_camera rs_rgbd.launch enable_gyro:=true enable_accel:=true`

RikisuT, 2024-12-30:
No change again
```Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Rikisu-2:41377/

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 310 lines cut ...]
 30/12 14:18:27,356 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,365 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,423 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,432 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,490 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,499 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,557 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,565 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,624 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,632 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:27,682 ERROR [139985947633408] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
 30/12 14:18:27,691 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[... cut ...]
 30/12 14:18:27,983 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:28,041 WARNING [139985721906944] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 30/12 14:18:28,160 ERROR [139985688336128] (error-handling.cpp:78) Error during polling error handler: get_xu(id=7) failed! Last Error: Success
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2024-12-30:
I note that the information at the top of this issue lists your librealsense version as 2.55.1, but your ROS launch is using 2.50.0.  Have you installed more than one librealsense version on the same computer, please?  Doing so could cause conflicts between the different versions.

RikisuT, 2024-12-30:
Oh sorry I had installed 2.50.0 but I had removed it 
I will check once again
Sorry

RikisuT, 2024-12-30:
1

RikisuT, 2024-12-30:
Marking as closed 
I haven't yet checked if I come across a problem I will reopen this issue

closed 2024-12-30 by RikisuT