realsenseai/librealsense #13643: Librealsense installation in raspberry pi

kkk2705, 2024-12-30:
I'm using raspberry pi 4 8gb kernel version 6.6
 I want to install librealsense so i used this link below https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
But mine kernel version is not suitable what to do

MartyG-RealSense, 2024-12-31:
Hi @kkk2705  Because of problems with using the librealsense SDK with kernels on Raspberry Pi, the best way to install the SDK on Pi is with the **libuvc backend** procedure at the link below, as this method of building from source code bypasses the kernel.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend procedure has a pre-made build script called **libuvc_installation.sh**.  If you want to build the Python wrapper at the same time as building librealsense then edit line 46 of the build script from this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

to this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS:bool=true`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

kkk2705, 2025-01-01:
will pyrealsense2 package support in python 3.11.2

MartyG-RealSense, 2025-01-01:
I do not have a time estimate for release of pyrealsense2 packages for Python 3.11 at the time of writing this.  However, **pip install pyrealsense2** package installation of the Arm version of the pyrealsense2 .whl packages does not work on Raspberry Pi boards, unfortunately.

The pyrealsense2 wrapper can be built for Python 3.11 from source code on Pi using the libuvc method above.

kkk2705, 2025-01-01:
ok thank you,
but method you suggestion I'm getting like
raspberrypi:~ $ python3 -c "import pyrealsense2 as rs; print(rs.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute '__version__'

kkk2705, 2025-01-01:
@MartyG-RealSense Hello Sir,

I followed the method you suggested for installation, and the RealSense Viewer is now working properly. However, I am facing an issue with pyrealsense2. When I run my code using pyrealsense2 in PyCharm, it shows "No device connected." Despite this, I can clearly see the camera feed when using cv2.VideoCapture(4) with OpenCV.

Could you please assist me in resolving this issue?

Thank you.

MartyG-RealSense, 2025-01-02:
Getting pyrealsense2 working on Pi can sometimes be complicated.  If you are using a Debian OS (Bookworm) and Python 3.11 then a guide at https://github.com/IntelRealSense/librealsense/issues/3062#issuecomment-1814386407 by a Pi user who also had Debian and 3.11 may be helpful.

You could also try running your Python script with sudo admin permissions.  For example, `sudo python3 test.py`

kkk2705, 2025-01-02:
Sir, 
They mentioned follow three path but I am not clear which one for what. Could you please tell for which files I need to add paths(python or library)
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/home/serveradmin/librealsense/build/Release

MartyG-RealSense, 2025-01-02:
If you are looking for the .bashrc file that you add the three above paths to the end of, it can be found in the Raspberry Pi home folder at the location **/home/pi/.bashrc**

If you are using a username on your Pi then it may be at `home/<username>/.bashrc`

kkk2705, 2025-01-02:
@MartyG-RealSense 
Thank you, sir. 
Could you please let me know which files and paths need to be added, whether to the library or the Python path?

MartyG-RealSense, 2025-01-03:
The log states that pyrealsense2 has been imported successfully, so the path is likely okay.

Does it make a difference if you change **import pyrealsense2 as rs** to `import pyrealsense2.pyrealsense2 as rs`

kkk2705, 2025-01-03:
@MartyG-RealSense sir,
when I use import pyrealsense2.pyrealsense2 as rs,
kabil@raspberrypi:~$ sudo /home/kabil/123/.venv/bin/python /home/kabil/123/check2.py
Traceback (most recent call last):
  File "/home/kabil/123/check2.py", line 1, in <module>
    import pyrealsense2.pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'

MartyG-RealSense, 2025-01-03:
Okay, thank you.  So **import pyrealsense2 as rs** is the correct instruction to use on your Pi.

https://github.com/IntelRealSense/librealsense/issues/10635 states that the problem with the attributes was bug-fixed with PR https://github.com/IntelRealSense/librealsense/pull/13079

This fix is only present in librealsense versions 2.56.1 and newer though.

kkk2705, 2025-01-03:
@MartyG-RealSense sir,
Will the same solution works for me
If yes, could you please explain what they mentioned and how to solve

MartyG-RealSense, 2025-01-03:
Getting pyrealsense2 to work on Raspberry Pi boards can be difficult unfortunately and there is not a single solution that works in all cases, so guarantees of success for a particular installation method cannot be provided.  

The librealsense SDK on Pi also has some known limitations.  The best that can typically be achieved is basic depth and rGB color streams, without post-processing filters, depth-color alignment or other CPU-intensive functions.  IMU motion data also often cannot be accessed on a Pi.

The Raspberry Pi hardware platform is not officially supported by RealSense, though the main librealsense SDK (except for the wrappers) can be set up for it quite easily.  The best way for Pi owners to find out whether librealsense or a wrapper such as pyrealsense2 or ROS will work on their board is therefore to try it and see.  I sincerely apologize for the inconvenience.

kkk2705, 2025-01-04:
@MartyG-RealSense
Ok sir thank you but
 https://github.com/IntelRealSense/librealsense/issues/13643#issuecomment-2569093745 I'm also using v2.56.1, so could you please guide how to do. May be if I'm fixing that bug, can solve my issue just try

kkk2705, 2025-01-04:
@MartyG-RealSense sir,
I copied the pyrealsense2.cpython-310-aarch64-linux-gnu.so file to the pyrealsense2 directory as follows:
sudo cp /home/kabil/librealsense_build/librealsense-master/build/release/pyrealsense2.cpython-310-aarch64-linux-gnu.so /usr/local/lib/python3.10/site-packages/pyrealsense2/.
Then, I added the path in .bashrc and made some changes in the PyCharm run configuration. It's working now.
Thank you so much for your support.

MartyG-RealSense, 2025-01-04:
It's great news that you achieved a solution.  Thanks so much for the update and for sharing the details of your method!

MartyG-RealSense, 2025-01-06:
Hi @kkk2705  Do you require further assistance with this case, please?  Thanks!

kkk2705, 2025-01-06:
@MartyG-RealSense sir,
Now its ok.
If any doubts, i will ask
Thankyou sir

closed 2025-01-06 by MartyG-RealSense