realsenseai/librealsense #13644: kinfu always fails update(), everything else works

Abdob, 2024-12-30:
I'm using Ubuntu 22 and realsense D435i on a x86 system.

I'm able to run all samples I've tried, but the kinfu one is always failing the update and resetting every iteration.

What would make it reset everytime and not build the point cloud?

MartyG-RealSense, 2024-12-31:
Hi @Abdob  Another RealSense user on Ubuntu 22.04 had the same resetting problem with rs-kinfu at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/22782163881747-3D-Body-Scanning-with-D455?page=1#community_comment_23214420457363

Does it make a difference if you comment out lines 257-259 of the rs-kinfu script so that the post-processing filters are not applied, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L257-L259

Abdob, 2024-12-31:
Hi @MartyG-RealSense 

commenting out the lines 257-259 didn't make a difference, commenting out line 289 did prevent the reset from happening although the update is still reporting a failure. There is nothing to try from the 3D body scanning post, although I'm not currently investigating this further in the moment I'll be happy to try any other suggestions. I was able to obtain a 3D model of an object I attempted to scan but would like to be able to scan miscellaneous items in a small environment, if the real-sense can do that.

Thanks,
Abdo

MartyG-RealSense, 2025-01-01:
If you are able to use commercial software, the RealSense-supporting program **LIPScan 3D** may be a suitable alternative to rs-kinfu as LIPScan can scan small objects rotating on a turntable.

https://www.lips-hci.com/lipscan-3d

Abdob, 2025-01-02:
Thanks for sharing this, I will contact them to see if they support the realsense D405 since I now aware that it has the most practical ideal range for our application.

MartyG-RealSense, 2025-01-12:
Hi @Abdob  Do you have an update about this case that you can provide, please?  Thanks!

Abdob, 2025-01-14:
Hi @MartyG-RealSense , I decided to proceed with the D405, this dual axis turn-table: 

https://www.amazon.com/gp/product/B0B9FVYRYX/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1

and a custom solution to combine point clouds

closed 2025-01-14 by Abdob

MartyG-RealSense, 2025-01-14:
Hi @Abdob  Thanks very much for the update about the solution that you chose and for sharing the link!