realsenseai/librealsense #13645: Color-depth alignment method for two pipelines

kummerrrrrr, 2024-12-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Linux Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.55.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I have tried countless times to use one pipe to enable both color and depth streams, but the program always reports an error, including when I modified the example provided in the SDK. So I used two pipes to enable color and depth streams respectively, but a new problem emerged. I found that the depth data acquisition was inaccurate. I tried to use:
rs2::align align_to_color(RS2_STREAM_COLOR);
frames = align_to_color.process(frames);
but it doesn't work. Can you provide a method to align the two pipes?

MartyG-RealSense, 2024-12-31:
Hi @kummerrrrrr  Your thinking for taking data from one pipeline and accessing it in a second pipeline in order to perform alignment is along the right lines theoretically, but on the couple of rare occasions when other RealSense users have attempted it in the past, it has not worked.

Was it the C++ **rs-align** SDK example that you modified, please?  Due to the way in which the SDK's examples are built, code from the examples may not work if copied into a new project because the new project will lack the required linkages to the librealsense library.

If the SDK examples are installed on your computer, is the original unmodified version of rs-align able to run correctly if you go to the **usr/local/bin** folder of Ubuntu to find the pre-built executable version of rs-align there and launch it?

kummerrrrrr, 2024-12-31:
`vision@vision:/usr/local/bin$ ./rs-align
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55cd1fe3c640, config:0x55cd1fe25240):
    xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
`
The part I modified: I added the depth stream to the rs-color.c example, and used the command line to compile and execute it. I found that the error thrown was the same as my own program. I just executed rs-align, which was not modified, and the result was also an error. Is this a case of my build failing? Or some library version mismatch?
So my program can only use two pipes to configure the color stream and depth stream, but I know this will still cause problems.

MartyG-RealSense, 2024-12-31:
There does not seem to be any problems in your combination of camera, librealsense version, Ubuntu version or firmware version.

rs-color.c is an example program in the "C" language rather than C++, so it would be difficult to modify to make it a depth-color alignment program by using a C++ code reference.

There is an official code reference about creating depth-color alignment in "C" language at the link below if you are programming in that language.

https://github.com/IntelRealSense/librealsense/discussions/12320

closed 2025-01-12 by MartyG-RealSense