realsenseai/librealsense #13648: The video recorded by Intel RealSense Viewer has a different depth value than the bag file called by Python

zhanyaoaaaaaa, 2025-01-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 }                                                  | 
| Firmware Version                   | (v2.55.1)                                                 | 
| Operating System & Version |   {Win (8.1/10)                                        | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                          |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                                | python                                                  | 
| Segment			         |  others                                                   | 

### Issue Description
When I used the Intel RealSense Viewer toolkit to record a bag file, I found that the depth value of a fixed pixel in the bag file that I called with Python code did not match the pixel displayed by the Viewer toolkit.
For example, I set the stereo module → post processing → temporary filter → filter smooth alpha to 0. When recording a bag file in Viewer, the depth value of a fixed pixel does not change over time. But when I called the bag file with Python code, I found that the depth value of the fixed pixel still changes over time, and the magnitude of this change is very large.
Excuse me, did the bag file not save the post processing result?
What should I do when using Python code to call a bag file? Never record directly in Python, it's inconvenient, just use Python code to call the bag file.

zhanyaoaaaaaa, 2025-01-02:
When I used Python code to call the bag file recorded by inter realsense vierwer, I found that all the parameters for post processing were fixed. In other words, although I changed the post-processing parameters during the recording of the inter realsense vierwer, the bag file did not save the parameters I changed. Is this a bug in the inter realsense vierwer?
The code is as follows.
`option = rs.option.filter_smooth_alpha
value = temporal_filter.get_option_value(option)
print(f"The value of {option} is {value}")
option = rs.option.filter_smooth_delta
value = temporal_filter.get_option_value(option)
print(f"The value of {option} is {value}")`

MartyG-RealSense, 2025-01-02:
The effects of post-processing filters are not saved to a bag file, just the raw stream data.  Once the bag file has been loaded in, the filters have to be applied to the data in real-time.

zhanyaoaaaaaa, 2025-01-02:
Can you explain that when I apply temporal filtering to a deep frame, the attribute of this frame changes from "pyrealsense2. pyrealsense2. depthframe" to "pyrealsense2. pyrealsense2. frame", causing the function get-distance to fail and report an error: AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_distance'。

In this situation, even if I apply temporal filter, I cannot proceed with subsequent data processing??????

May I ask if there is a bug in the program?

zhanyaoaaaaaa, 2025-01-02:
Or alternatively, can I change the parameters of the time filter when recording files in Python, so that the depth value of the bag file recorded in Python changes? If not, why is post-processing necessary? Because I found that the default post-processing parameters do not match the actual results in processing depth values.

Please reply to me as soon as possible, as the setting of this post-processing parameter has been bothering me for a long time and urgently needs to be resolved now!

zhanyaoaaaaaa, 2025-01-02:
Or alternatively, can I change the parameters of the temporal filter when recording files in Python, so that the depth value of the bag file recorded in Python changes? If not, why is post-processing necessary? Because I found that the default post-processing parameters do not match the actual results in processing depth values.

Please reply to me as soon as possible, as the setting of this post-processing parameter has been bothering me for a long time and urgently needs to be resolved now!

Looking forward to your reply,thanks!

closed 2025-01-12 by MartyG-RealSense