realsenseai/librealsense #13651: Installing d455f for Nvidia Isaac_ros but not detecting the device

iine4649, 2025-01-02:
| Camera Model                       | D455f | 
| Firmware Version                   | 2.55.1 | 
| Operating System & Version | Ubuntu | 
| Kernel Version (Linux Only)    | Ubuntu 22.04.5 LTS | 
| Platform                                 | NVIDIA Jetson Orin Nano |

Hi. I installed and related packages correctly following directions [here](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/realsense_setup.html) But it gives me that no realsense device is detected
And there is no camera view. Just x,y,z axis are appear. I ran 
`rs-enumerate-devices` 
but it gave me this:
`No device detected. Is it plugged in?`

I tried 
`lsusb`
gave me following:
`
Bus 002 Device 002: ID 0bda:0489 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 005: ID 8086:0b5c Intel Corp. Intel(R) RealSense(TM) Depth Camera 455f
Bus 001 Device 004: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse
Bus 001 Device 002: ID 0bda:5489 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

 I am developing ros_visual_slam on Nvidia Jetson Orin nano for the purpose of scanning 3d image

MartyG-RealSense, 2025-01-03:
Hi @iine4649  Are you following the guide's advice to use camera firmware version 5.13.0.50 please?  If you are, support for D455f was not added in the firmware until the later version 5.15.0.2.  So firmware 5.13.0.50 would not be able to detect a D455f.

If you are using SDK 2.55.1 as the guide advises then the recommended firmware to use with it is **5.16.0.1**

iine4649, 2025-01-04:
Thanks for the reply @MartyG-RealSense 

It seems that the Firmware Update Tool is not working either, 
`rs-fw-update -l`
ends up with 
`There are no connected devices`

JetPack version is 6.1

I suspected this was the case for udev, but it seemed fine to me

Thanks!

MartyG-RealSense, 2025-01-04:
RealSense camera models that are equipped with an IMU, like D455f, may not be detected by JetPack 6.  This is because JP6 removed a component called **hiddraw** that the RealSense IMU depends upon.

You likely cannot downgrade to JetPack 5.1.3 that works with the IMU if you are using Ubuntu 22.04.  So an alternative fix is to build the librealsense SDK from source code with the **libuvc backend** installation method, which bypasses the kernel and so can help to resolve problems related to the JetPack version.

The website **JetsonHacks** provide a pre-made build script called buildLibrealsense.sh for building the SDK from source code on Jetson with this method.

https://github.com/JetsonHacksNano/installLibrealsense?tab=readme-ov-file#buildlibrealsensesh

MartyG-RealSense, 2025-01-12:
Hi @iine4649  Do you require further assistance with this case, please?  Thanks!

iine4649, 2025-01-15:
Hi @MartyG-RealSense Thank you sooo much! it has finally worked out!

MartyG-RealSense, 2025-01-15:
That's great to hear, @iine4649 - thanks very much for the update!

closed 2025-01-20 by MartyG-RealSense