realsenseai/librealsense #13653: In Android, how to set the initial depth camera resolution 1280x720 or 1920x1080 ？

1programmer2dfg3, 2025-01-06:
I am currently setting through this function
`config.enableStream(StreamType.DEPTH, 1280, 720);`
The result is abnormal, camera data cannot stream in, preview black screen.
But setting 640x480 is no problem。

2025-01-06 10:18:27.365 27702-27702 librs                   com.gather.terminal                  I  ... /dev/bus/usb/001/003
2025-01-06 10:18:27.365 27702-27702 librs                   com.gather.terminal                  I  Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
2025-01-06 10:18:27.366 27702-27702 librs                   com.gather.terminal                  I  ... /dev/bus/usb/001/003
2025-01-06 10:18:27.366 27702-27702 librs                   com.gather.terminal                  I  Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
2025-01-06 10:18:27.367 27702-27702 RemoteCaptureFragment   com.gather.terminal                  D  try start streaming
2025-01-06 10:18:27.367 27702-27702 librs                   com.gather.terminal                  I  ... /dev/bus/usb/001/003
2025-01-06 10:18:27.367 27702-27702 librs                   com.gather.terminal                  I  Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
2025-01-06 10:18:27.390 27702-27715 gather.termina          com.gather.terminal                  I  ThreadFlipBegin blocked for 11.482ms
2025-01-06 10:18:27.404 27702-27702 librs                   com.gather.terminal                  I  Request: Z16 Depth, 
                                                                                                    Resolved to: Z16 Depth, 
2025-01-06 10:18:27.404 27702-28865 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-06 10:18:27.405 27702-28865 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-06 10:18:27.406 27702-28865 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-06 10:18:27.407 27702-28865 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-06 10:18:27.407 27702-28865 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-06 10:18:27.408 27702-28865 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-06 10:18:27.408 27702-27702 librs                   com.gather.terminal                  E  Probe-commit control transfer failed with error: RS2_USB_STATUS_OTHER
2025-01-06 10:18:27.411 27702-27702 librs                   com.gather.terminal                  E  [rs2_pipeline_start_with_config( pipe:0x7273c61290, config:0x7273c707b0 ) Invalid Value] 
                                                                                                    Failed to resolve the request: 
                                                                                                     Format: Z16, width: 1280, height: 720
                                                                                                    
                                                                                                    Into:
                                                                                                     Formats: 
                                                                                                      Z16

MartyG-RealSense, 2025-01-06:
Hi @1programmer2dfg3  Please try adding the FPS speed to your Android stream configuration instruction.  The **Failed to resolve the request** error could occur if the program is automatically attempting to use 30 FPS but the camera is unable to provide it.  A good FPS value to test with is '6', or '5' if you have the RealSense D455 camera model.

`cfg.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 6);`

1programmer2dfg3, 2025-01-10:
Thank you for your reply. I have tested it and found that it did not take effect. The screen I preview is still missing. My test was passed by one, which is

**`config.enableStream(StreamType.DEPTH, 1,1280, 720, StreamFormat.Z16, 6)`**

Below is the error log, help me look again, thank you

2025-01-10 15:01:17.247 29447-29447 librs                   com.gather.terminal                  E  [rs2_pipeline_wait_for_frames( pipe:0x7273c630f0 ) Wrong Api Call Sequence] wait_for_frames cannot be called before start()
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  java.lang.RuntimeException: wait_for_frames cannot be called before start()
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at com.intel.realsense.librealsense.Pipeline.nWaitForFrames(Native Method)
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at com.intel.realsense.librealsense.Pipeline.waitForFrames(Pipeline.java:45)
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at com.intel.realsense.librealsense.Pipeline.waitForFrames(Pipeline.java:41)
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at com.gather.terminal.ui.fragment.RemoteCaptureFragment$3.run(RemoteCaptureFragment.java:374)
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at android.os.Handler.handleCallback(Handler.java:938)
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at android.os.Handler.dispatchMessage(Handler.java:99)
2025-01-10 15:01:17.247 29447-29447 System.err              com.gather.terminal                  W  	at android.os.Looper.loopOnce(Looper.java:201)
2025-01-10 15:01:17.377 29447-29447 System.err              com.gather.terminal                  W  	at android.app.ActivityThread.main(ActivityThread.java:7870)
2025-01-10 15:01:17.377 29447-29447 System.err              com.gather.terminal                  W  	at java.lang.reflect.Method.invoke(Native Method)
2025-01-10 15:01:17.377 29447-29447 System.err              com.gather.terminal                  W  	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
2025-01-10 15:01:17.377 29447-29447 System.err              com.gather.terminal                  W  	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

1programmer2dfg3, 2025-01-10:
**this is error log:**
2025-01-10 15:12:45.574 23164-23164 librs                   com.gather.terminal                  E  [rs2_pipeline_wait_for_frames( pipe:0x7273c6d1b0 ) Wrong Api Call Sequence] wait_for_frames cannot be called before start()
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  java.lang.RuntimeException: wait_for_frames cannot be called before start()
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at com.intel.realsense.librealsense.Pipeline.nWaitForFrames(Native Method)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at com.intel.realsense.librealsense.Pipeline.waitForFrames(Pipeline.java:45)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at com.intel.realsense.librealsense.Pipeline.waitForFrames(Pipeline.java:41)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at com.gather.terminal.ui.fragment.RemoteCaptureFragment$3.run(RemoteCaptureFragment.java:375)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at android.os.Handler.handleCallback(Handler.java:938)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at android.os.Handler.dispatchMessage(Handler.java:99)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at android.os.Looper.loopOnce(Looper.java:201)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at android.os.Looper.loop(Looper.java:288)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at android.app.ActivityThread.main(ActivityThread.java:7870)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at java.lang.reflect.Method.invoke(Native Method)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
2025-01-10 15:12:45.574 23164-23164 System.err              com.gather.terminal                  W  	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
2025-01-10 15:12:45.574 23164-23164 librs                   com.gather.terminal                  I  ... /dev/bus/usb/001/003
2025-01-10 15:12:45.574 23164-23164 librs                   com.gather.terminal                  I  Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
2025-01-10 15:12:45.575 23164-23164 librs                   com.gather.terminal                  I  Found 107 unreachable streams, 107 extrinsics deleted
2025-01-10 15:12:45.644 23164-23164 librs                   com.gather.terminal                  I  Request: Z16 Depth, 
                                                                                                    Resolved to: Z16 Depth, 
2025-01-10 15:12:45.644 23164-24877 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-10 15:12:45.645 23164-24877 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-10 15:12:45.645 23164-24877 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-10 15:12:45.646 23164-24877 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-10 15:12:45.647 23164-24877 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-10 15:12:45.654 23164-24877 librs                   com.gather.terminal                  W  control_transfer returned error, index: 1, error: Broken pipe, number: 32
2025-01-10 15:12:45.656 23164-23164 librs                   com.gather.terminal                  E  Probe-commit control transfer failed with error: RS2_USB_STATUS_OTHER
2025-01-10 15:12:45.669 23164-23164 librs                   com.gather.terminal                  E  [rs2_pipeline_start_with_config( pipe:0x7273c6d1b0, config:0x7273c86fb0 ) Invalid Value] 
                                                                                                    Failed to resolve the request: 
                                                                                                    	Format: Z16, width: 1280, height: 720
                                                                                                    
                                                                                                    Into:
                                                                                                    	Formats: 
                                                                                                    	 Z16

1programmer2dfg3, 2025-01-12:
848x480 is also OK, this is the configuration code
`config.enableStream(StreamType.DEPTH, 848, 480);`
But 1280x720 is no good, I need to set this, what should I do

MartyG-RealSense, 2025-01-12:
What RealSense camera model are you using, please?

1programmer2dfg3, 2025-01-12:
Inter D435

1programmer2dfg3, 2025-01-12:
Intel D435

MartyG-RealSense, 2025-01-12:
Ok, so 1280x720 should be achievable as it is supported on D435.  

Are you using the official USB cable supplied with the camera or a cable that you chose yourself?

1programmer2dfg3, 2025-01-12:
Using the official original

MartyG-RealSense, 2025-01-12:
Thank you.  And what is your librealsense SDK version and camera firmware driver version, please?

1programmer2dfg3, 2025-01-12:
SDK version is 2.55.1
This api gets it 
`RsContext.getVersion()`
and How do I check the camera firmware driver version,please?

MartyG-RealSense, 2025-01-12:
If you can use the camera on PC then you can retrieve the firmware version with the command **rs-enumerate-devices -s**

Alternatively, if you can access the realsense-viewer tool then clicking the 'Info' button at the top of the options side-panel will list the firmware.  For librealsense 2.55.1, firmware version **5.16.0.1** should be installed in the camera.

1programmer2dfg3, 2025-01-13:
At present, the camera is plugged into the Android tablet and cannot be connected to the PC. How can I check the firmware version of the camera?

1programmer2dfg3, 2025-01-13:
[image]
I plug the usb of the camera into the computer and check through this software. The information I see now is in the picture. What do I need to do to support the 1280x720 resolution image ？

1programmer2dfg3, 2025-01-13:
[image]
firmware version 5.16.0.1

MartyG-RealSense, 2025-01-13:
I cannot see anything in the images that would prevent 1280x720 resolution from working, and you are using the correct 5.16.0.1 firmware for SDK version 2.55.1.
 
When you open the resolution drop-down menu whilst the Stereo Module is turned off to choose 1280x720 and then open the FPS speed drop-down and select 30, does the resolution stay as 1280x720 or change to a different resolution setting?

For example, if I choose 1280x720 and then select 60 FPS for it:

[image]

the resolution setting changes to 640x480 at 60 FPS because 1280x720 does not support use of 60 FPS.

[image]

closed 2025-01-26 by MartyG-RealSense