realsenseai/librealsense #13655: Why didn't the "MATLAB Developer Package" option appear when I installed SDK versions 2.56.3 and 2.56.2?

Xk2033106447, 2025-01-06:
Why didn't the "MATLAB Developer Package" option appear when I installed SDK versions 2.56.3 and 2.56.2?

MartyG-RealSense, 2025-01-06:
Hi @Xk2033106447  The MATLAB developer package is not included as an installation option after librealsense SDK version 2.54.1.  You can obtain it by installing 2.54.1 on Windows by downloading and running the **Intel.RealSense.SDK-WIN10** installer program from the 'Assets' file list at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

When a newer librealsense version is installed, the MATLAB developer package will still be present on your computer.

[image]

The MATLAB developer package usually works fine with the latest SDK versions.

As you already have a beta SDK version installed, what might work best for you when using the 2.54.1 installer program is to untick all options except the MATLAB component.

Xk2033106447, 2025-01-06:
Thank you very much, I will try your program!

Xk2033106447, 2025-01-06:
> 嗨，在 librealsense SDK 版本 2.54.1 之后，MATLAB 开发人员包未作为安装选项包含在内。您可以通过在 Windows 上安装 2.54.1 来获取它，方法是从下面链接的“资产”文件列表中下载并运行**英特尔.RealSense.SDK-WIN10** 安装程序。
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1
> 
> 安装较新的 librealsense 版本后，MATLAB 开发人员包仍将存在于您的计算机上。
> 
> ![图像](https://private-user-images.githubusercontent.com/41145062/400431665-22a7f9c0-b5be-4e97-9ec7-4f02adc94a28.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzA4MTgsIm5iZiI6MTczNjE3MDUxOCwicGF0aCI6Ii80MTE0NTA2Mi80MDA0MzE2NjUtMjJhN2Y5YzAtYjViZS00ZTk3LTllYzctNGYwMmFkYzk0YTI4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA2VDEzMzUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIyOWYyMGNhYWNiZjUwNTUxZDk1NGI3NWRiMzBlY2FhYWQ2NWM5ZjBjNDQ4ZDg3MDQ5MjgxOTY3NDUyYzZkYWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.RL73JLivwNQWYCOK70-u5nMIYaJDgpzBnyAe-9ulfac)
> 
> MATLAB 开发人员包通常适用于最新的 SDK 版本。
> 
> 由于您已经安装了 Beta 版 SDK，因此在使用 2.54.1 安装程序时，最适合您的方法是取消选中自定义安装框中除 MATLAB 组件之外的所有组件。

Thank you very much, I will try your program!

closed 2025-01-20 by MartyG-RealSense