realsenseai/librealsense #13660: Clarification reset factory

mockup-partner, 2025-01-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.7 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.55.1                       | 
| Language                            |  viewer / python                          | 
| Segment			| Robot | 

### Issue Description

Hi, 

I am currently investigating the use of calibration tools provided by intel such as on-chip calibration, focal imbalance and tare. I have some questions related to that topic. 

- when I run on chip calibration in realsense-viewer and apply the change, if I compare the calibration data I don't notice any difference. This seems to be expected as suggested [here](https://github.com/IntelRealSense/librealsense/issues/10638#issuecomment-1172342235). Is it possible to have a clarification on why it is not expected to change ? Is it because the changes are so small that with the precision displayed in this table values won't change, or is it because the on chip calibration optimizes parameters not exposed in the calibration table or is it something else ? 
- a related question is how can I programmatically compare the calibration data before and after calibration to check what changed ? Ideally I would like to do it in python 
- a final question, could you clarify the difference between hardware reset and reset factory in the realsense-viewer ? 

Thank you in advance

MartyG-RealSense, 2025-01-08:
Hi @mockup-partner  Thanks very much for your questions.

1.  If you have saved the updated calibration to the camera hardware after on-chip calibration has been completed then I would recommend comparing intrinsics and extrinsics before and after calibration by using the calibration information mode of the **rs-enumerate-devices** tool instead of using the Calibration Data panel of realsense-viewer.  You can launch rs-enumerate-devices in calibration info mode with the command below: 
 
`rs-enumerate-devices -c`

2.  The current intrinsic and extrinsic values of the camera can be queried in Python with the instructions **get_intrinsics()** and **get_extrinsics_to()**.  https://github.com/IntelRealSense/librealsense/issues/3986 has Python scripting for retrieving and prnting intrinsics with get_intrinsics() whilst https://github.com/IntelRealSense/librealsense/issues/10252 demonstrates retrieving and printing extrinsic values with get_extrinsics_to().

3.  Hardware reset disconnects and reconnects the camera, similar to when the camera is unplugged and then plugged back in.  A factory reset does not restart the camera, but instead changes the current calibration values of the camera to the ones that it had when it was brand new and first purchased.

mockup-partner, 2025-01-08:
Thanks @MartyG-RealSense for your answer. I am closing the issue.

closed 2025-01-08 by mockup-partner