realsenseai/librealsense #13662: realsense-viewer not getting installed in Docker Container

rdlynx19, 2025-01-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Docker Container with Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.9.3-76060903-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |   Unable to find in Docker container?                          | 
| Language                            | N/A                      | 
| Segment			|  ROS2/ Robotics                          | 

### Issue Description
Hello, 
I have a ROS2 Humble and Ubuntu 22.04 Docker container on VS code dev environment. I am trying to setup the realsense camera within the container. I have tried to follow all the instructions to install realsense SDK and librealsense in the Docker. 
This is my Docker File
```
FROM althack/ros2:humble-gazebo 

# ** [Optional] Uncomment this section to install additional packages. **
#
ENV DEBIAN_FRONTEND=noninteractive
# RUN apt-get update \
#    && apt-get -y install --no-install-recommends <your-package-list-here> \
#    #
#    # Clean up
#    && apt-get autoremove -y \
#    && apt-get clean -y \
#    && rm -rf /var/lib/apt/lists/*
# ENV DEBIAN_FRONTEND=dialog

# Set up auto-source of workspace for ros user
ARG WORKSPACE
RUN echo "if [ -f ${WORKSPACE}/install/setup.bash ]; then source ${WORKSPACE}/install/setup.bash; fi" >> /home/ros/.bashrc

RUN mkdir -p /etc/apt/keyrings && curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/keyrings/librealsense.pgp > /dev/null
RUN echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | sudo tee /etc/apt/sources.list.d/librealsense.list

RUN apt-get update \
 && apt-get install -y --no-install-recommends \
    build-essential \
    cmake \
    git-all \
    libusb-1.0-0 \
    udev \
    software-properties-common \
 && rm -rf /var/lib/apt/lists/*
 
RUN apt-get update \
 && apt-get install -y \ 
    ros-humble-librealsense2* \
    ros-humble-realsense2-* \
 && rm -rf /var/lib/apt/lists/*

RUN apt-get update \
 && apt-get install -y ros-humble-rtabmap-ros \
 && apt-get install -y ros-humble-turtlebot3* \
 && apt-get install -y usbutils \
 && rm -rf /var/lib/apt/lists/*

RUN /bin/bash -c "source /opt/ros/humble/setup.bash"

ENV DEBIAN_FRONTEND=dialog

CMD [ "rs-enumerate-devices", "--compact" ]
```
I am using the VS Code Dev environment so the Docker container is run using a `devcontainer.json` file like this.
```
// See https://aka.ms/vscode-remote/devcontainer.json for format details.
{	"name" : "athackst-humble",
	"dockerFile": "Dockerfile",
	"build": {
		"args": {
			"WORKSPACE": "${containerWorkspaceFolder}"
		}
	},
	"remoteUser": "ros",
	"runArgs": [
		"--privileged",
		"--network=host",
		"--cap-add=SYS_PTRACE",
		"--security-opt=seccomp:unconfined",
		"--security-opt=apparmor:unconfined",
		"--volume=/tmp/.X11-unix:/tmp/.X11-unix",
		"--volume=/mnt/wslg:/mnt/wslg",
		"--ipc=host",
		"--device=/dev/bus/usb:/dev/bus/usb"
		// uncomment to use intel iGPU
		// "--device=/dev/dri"
	],
	"containerEnv": {
		"DISPLAY": "${localEnv:DISPLAY}", // Needed for GUI try ":0" for windows
		"WAYLAND_DISPLAY": "${localEnv:WAYLAND_DISPLAY}",
		"XDG_RUNTIME_DIR": "${localEnv:XDG_RUNTIME_DIR}",
		"PULSE_SERVER": "${localEnv:PULSE_SERVER}",
		"LIBGL_ALWAYS_SOFTWARE": "1" // Needed for software rendering of opengl
	},
	// Set *default* container specific settings.json values on container create.
	"customizations": {
		"vscode": {
			"extensions": [
				"althack.ament-task-provider",
				"betwo.b2-catkin-tools",
				"DotJoshJohnson.xml",
				"ms-azuretools.vscode-docker",
				"ms-iot.vscode-ros",
				"ms-python.python",
				"ms-vscode.cpptools",
				"redhat.vscode-yaml",
				"smilerobotics.urdf",
				"streetsidesoftware.code-spell-checker",
				"twxs.cmake",
				"yzhang.markdown-all-in-one",
				"zachflower.uncrustify"
			]
		}
	}
}
```

I know the librealsense packages for ROS2 are installed.
```
ros@tokyo:/workspaces/docker_ros2_vscode$ dpkg -l | grep "realsense"
ii  ros-humble-librealsense2                           2.55.1-1jammy.20241125.233100           amd64        Library for controlling and capturing data from the Intel(R) RealSense(TM) D400 devices.
ii  ros-humble-librealsense2-dbgsym                    2.55.1-1jammy.20241125.233100           amd64        debug symbols for ros-humble-librealsense2
ii  ros-humble-realsense2-camera                       4.55.1-1jammy.20241128.030800           amd64        RealSense camera package allowing access to Intel D400 3D cameras
ii  ros-humble-realsense2-camera-dbgsym                4.55.1-1jammy.20241128.030800           amd64        debug symbols for ros-humble-realsense2-camera
ii  ros-humble-realsense2-camera-msgs                  4.55.1-1jammy.20241128.011951           amd64        RealSense camera_msgs package containing realsense camera messages definitions
ii  ros-humble-realsense2-camera-msgs-dbgsym           4.55.1-1jammy.20241128.011951           amd64        debug symbols for ros-humble-realsense2-camera-msgs
ii  ros-humble-realsense2-description                  4.55.1-1jammy.20241128.033038           amd64        RealSense description package for Intel 3D D400 cameras
```

But typing `realsense-viewer` returns `command not found`
Another important thing to note is that I can get the docker container to access the realsense camera only by setting the `privileged` flag as true. 

I don't understand if there is something wrong in my installation steps, or am I running the container with the incorrect arguments.

Any help will be appreciated. Thank you.

MartyG-RealSense, 2025-01-08:
Hi @rdlynx19  Most successful RealSense Docker installations have the `privileged` flag set as true, so that is normal and correct.

It looks as though you are using the instructions from the **distribution_linux.md** page in your Docker file sign in to the package keyserver.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

But then librealsense main package `librealsense2-dkms` and the `librealsense2-utils` package, which installs examples and tools such as realsense-viewer, are not installed.  Instead, librealsense and the ROS wrapper are apparently installed using the packages on the ROS server, `ros-humble-librealsense2` and `ros-humble-realsense2`.  

If you install the librealsense SDK from the packages on the ROS server then realsense-viewer is not included in the installation, which may be why your realsense-viewer launch command is not recognized.

To use realsense-viewer, I recommend installing librealsense with the packages librealsense-dkms and librealsense-utils using th einstructions on the distribution_linux.md page.  Then install the ROS wrapper from the ROS server with `ros-humble-realsense2`

rdlynx19, 2025-01-08:
Hi @MartyG-RealSense 
Thanks for your comment. I tried to add the lines based on **distribution_linux.md** in my docker file
The docker file now looks like
```
FROM althack/ros2:humble-gazebo 

# ** [Optional] Uncomment this section to install additional packages. **
#
ENV DEBIAN_FRONTEND=noninteractive
# RUN apt-get update \
#    && apt-get -y install --no-install-recommends <your-package-list-here> \
#    #
#    # Clean up
#    && apt-get autoremove -y \
#    && apt-get clean -y \
#    && rm -rf /var/lib/apt/lists/*
# ENV DEBIAN_FRONTEND=dialog

# Set up auto-source of workspace for ros user
ARG WORKSPACE
RUN echo "if [ -f ${WORKSPACE}/install/setup.bash ]; then source ${WORKSPACE}/install/setup.bash; fi" >> /home/ros/.bashrc

RUN mkdir -p /etc/apt/keyrings && curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/keyrings/librealsense.pgp > /dev/null
RUN echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | sudo tee /etc/apt/sources.list.d/librealsense.list

RUN apt-get update \
 && apt-get install -y --no-install-recommends \
    build-essential \
    cmake \
    git-all \
    libusb-1.0-0 \
    udev \
    software-properties-common \
 && rm -rf /var/lib/apt/lists/*

# Installing realsense-viewer
RUN apt-get update \
 && apt-get install -y --no-install-recommends \
    librealsense2-dkms \
    librealsense2-utils \ 
 && rm -rf /var/lib/apt/lists/*
 
RUN apt-get update \
 && apt-get install -y \ 
   #  ros-humble-librealsense2* \
    ros-humble-realsense2-* \
 && rm -rf /var/lib/apt/lists/*

RUN apt-get update \
 && apt-get install -y ros-humble-rtabmap-ros \
 && apt-get install -y ros-humble-turtlebot3* \
 && apt-get install -y usbutils \
 && rm -rf /var/lib/apt/lists/*

RUN /bin/bash -c "source /opt/ros/humble/setup.bash"

ENV DEBIAN_FRONTEND=dialog

CMD [ "rs-enumerate-devices", "--compact" ]
```
However, the docker container build now fails.
The logs from VS Code Remote Container is attached here.

[remoteContainers-2025-01-08T19-38-06.036Z.log](https://github.com/user-attachments/files/18352318/remoteContainers-2025-01-08T19-38-06.036Z.log)

The error is indicated in lines **3135-3180**
```

17.01 invoke-rc.d: could not determine current runlevel

17.01 invoke-rc.d: policy-rc.d denied execution of start.

17.08 Setting up gcc-12 (12.3.0-1ubuntu1~22.04) ...

17.12 Setting up libpango1.0-dev:amd64 (1.50.6+ds-2ubuntu1) ...

17.16 Setting up libgtk-3-dev:amd64 (3.24.33-1ubuntu2.2) ...

17.19 Setting up librealsense2-utils:amd64 (2.55.1-0~realsense.12474) ...

17.23 Setting up dkms (2.8.7-2ubuntu2.2) ...

17.55 Setting up librealsense2-dkms (1.3.28-0ubuntu1) ...

17.61 Loading new librealsense2-dkms-1.3.28 DKMS files...

17.75 It is likely that 6.9.3-76060903-generic belongs to a chroot's host

17.76 Building for 

17.76 

17.76 Loading the modified modules into kernel...dpkg: error processing package librealsense2-dkms (--configure):

17.77  installed librealsense2-dkms package post-installation script subprocess returned error exit status 1

17.77 Processing triggers for libc-bin (2.35-0ubuntu3.8) ...

17.88 Errors were encountered while processing:

17.88  librealsense2-dkms
[2025-01-08T19:38:24.772Z] 17.91 E: Sub-process /usr/bin/dpkg returned an error code (1)
------
[2025-01-08T19:38:24.773Z] Dockerfile-with-features:34
--------------------
  33 |     
  34 | >>> RUN apt-get update \
  35 | >>>  && apt-get install -y --no-install-recommends \
  36 | >>>     librealsense2-dkms \
  37 | >>>     librealsense2-utils \ 
  38 | >>>  && rm -rf /var/lib/apt/lists/*
  39 |      
--------------------
[2025-01-08T19:38:24.773Z] ERROR: failed to solve: process "/bin/sh -c apt-get update  && apt-get install -y --no-install-recommends     librealsense2-dkms     librealsense2-utils  && rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 100
```

MartyG-RealSense, 2025-01-09:
Another RealSense user who experienced **Errors were encountered while processing** when attempting to install `librealsense2-dkms` from a Docker file shared their solution at https://github.com/IntelRealSense/librealsense/issues/13536#issuecomment-2495419513

MartyG-RealSense, 2025-01-10:
You are very welcome.  It's excellent news that you achieved a solution.  Thanks very much for the update!

closed 2025-01-20 by MartyG-RealSense