realsenseai/librealsense #13664: Realsense Support for AGX Orin with Jetpack 6.1

shreshtashetty, 2025-01-08:
I have tried installing the Realsense SDK first through the linux drivers, then through the [libuvc `.sh`](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) file, but the Realsense Viewer is not able to detect the camera. lsusb shows that there is a Realsense camera attached, though.

System: NVidia Jetson AGX Orin.
Ubuntu Version: 22.04
Jetpack Version: 6.1
Camera: D455.

Also, after running the libuvc .`sh` file, the Realsense Viewer shows the following message:-
```
Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

```
Any fixes for Jetpack 6.1 without downgrading to Jetpack 5.1?
Also, any idea if or when a new SDK/MIPI platform driver/libuvc installation for Jetpack 6.1 will be released?

MartyG-RealSense, 2025-01-08:
Hi @shreshtashetty  The **Multiple realsense udev-rules were found!** message usually occurs if librealsense has been installed from both packages and source code on the same computer.  It can be fixed easily by deleting the rule at **/etc/udev/rules.d/99-realsense-libusb.rules**.  After the deletion, the message should disappear.  

The /lib rule should not be deleted, as doing so breaks linkages to DKMS packages.

There is not currenty information available about timescales regarding future support of specific JetPack versions.

MartyG-RealSense, 2025-01-20:
Hi @shreshtashetty  Do you have an update about this case that you can provide, please?  Thanks!

shreshtashetty, 2025-01-21:
Hi, this happened because I had installed Realsense Drivers through both the libuvc.sh method as well as the Debian Package.
When I uninstalled the Debian installation, I stopped getting the message, "Multiple realsense udev-rules were found!", but the problem of the Realsense Viewer not showing the camera images and motion module persisted. 
Downgrading to Jetpack 6.0 solved the problem. Jetpack 6.0 + the libuvc.sh installation made everything work fine.

MartyG-RealSense, 2025-01-21:
I'm pleased to hear that you achieved a solution!  As the problem is fixed, I will close this issue.  Thanks very much for the update!

closed 2025-01-21 by MartyG-RealSense

Dharshan-BL, 2025-02-09:
Hi, I just wanted to add mention the following for others in the future: that with Jetson AGX orin with `Jetpack 6.2` + `libuvc.sh` installation (librealsense2 v2.55.1) + building `ros-wrapper` (V4.55.1) is stable on ROS2 humble. (Additionally, I use `D435i` and I had to downgrade my firmware to V5.13.0.50).

MartyG-RealSense, 2025-02-09:
Thanks so much @Dharshan-BL for sharing the details of the configuration that worked for you!