realsenseai/librealsense #13666: No device connected with .py file but Run in realsense-viewer

amaraashishmahatokushwaha, 2025-01-08:
### No device connected while running .py file
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455f 
| Firmware Version                   | 5.16.0.1
| Operating System & Version |   UBUNTU 22.04
| Kernel Version (Linux Only)    |  5.15.136-tegra                                      
| Platform                                 | Jetson ORIN NANO|
| SDK Version                            |  2.55.1                         

### Issue Description
I installed sdk using libuvc_installation.sh  . So it installed sdk version 2.22.1 ( i know while opening realsense-viewer) . 
"""""""""""""I will list the problems""""""
1)Realsense-viewer works, but third option: motion module is not working
2) I installed pyrealsense2 using pip and when  i run .py file such as to show depth images and all , in that it shows:
orin@orin-desktop:~/Downloads$ python3 main_collision.py 
Traceback (most recent call last):
  File "/home/orin/Downloads/main_collision.py", line 102, in <module>
    main()
  File "/home/orin/Downloads/main_collision.py", line 97, in main
    node = CollisionAvoidanceNode()
  File "/home/orin/Downloads/main_collision.py", line 21, in __init__
    self.pipeline.start(self.config)

**RuntimeError: No device connected**

I have gone through the issues and read all the solutions available such as using installation process steps using libuvc, insert cable fast in jetson( so that jetson detects as USB 3.2). But nothing works and solved my problem. 

I tried to install realsense-ros from source from official documentation and when i run this launch file it shows:
**orin@orin-desktop:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py**
[INFO] [launch]: All log files can be found below /home/orin/.ros/log/2025-01-08-15-36-44-194901-orin-desktop-54617
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [54618]
[realsense2_camera_node-1] [INFO] [1736330804.529719337] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1736330804.529903730] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736330804.529941556] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1736330805.110539990] [camera.camera]: Device with serial number 311322301523 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1736330805.110748575] [camera.camera]: Device with physical ID 2-1.3-4 was found.
[realsense2_camera_node-1] [INFO] [1736330805.110776289] [camera.camera]: Device with name Intel RealSense D455F was found.
[realsense2_camera_node-1] [INFO] [1736330805.111187956] [camera.camera]: Device with port number 2-1.3 was found.
[realsense2_camera_node-1] [INFO] [1736330805.111220725] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1736330805.111400606] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1736330805.111840594] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1736330805.111880340] [camera.camera]: Device Name: Intel RealSense D455F
[realsense2_camera_node-1] [INFO] [1736330805.111904149] [camera.camera]: Device Serial No: 311322301523
[realsense2_camera_node-1] [INFO] [1736330805.111920086] [camera.camera]: Device physical port: 2-1.3-4
[realsense2_camera_node-1] [INFO] [1736330805.111933911] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1736330805.111949112] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1736330805.111962776] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1736330809.753840573] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1736330809.755065974] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1736330810.736623679] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1736330810.738904073] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1736330810.739067857] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1736330810.744685046] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736330810.744887263] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736330810.861979334] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1736330810.941118088] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  08/01 15:36:50,943 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:50,955 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:50,968 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:50,979 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:50,991 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:51,003 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:51,014 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:51,026 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:36:51,037 WARNING [281473178069216] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... 70 lines cut ...]
[realsense2_camera_node-1]  08/01 15:43:18,063 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1736331198.076980996] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1736331198.108841811] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1736331198.111330535] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  08/01 15:43:18,585 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:43:18,596 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:43:18,607 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:43:18,619 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:43:18,632 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  08/01 15:43:18,948 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  08/01 15:43:18,959 WARNING [281472292350176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
5.16.0.1

even i am not able to read imu data and all. My main problem is to run .py file .

@MartyG-RealSense Thankyou for help in advance.

MartyG-RealSense, 2025-01-08:
Hi @amaraashishmahatokushwaha  Is the JetPack version that you are using with your Jetson and Ubuntu 22.04 **JetPack 6** please?  If it is then the IMU data may not be provided or the camera may not detect at all.  This is because JP6 removed a function called **hiddraw** that RealSense cameras equipped with an IMU depend upon (as IMU-equipped RealSense cameras are known as HID devices).

Building the librealsense SDK from source code with the libuvc backend can provide IMU data in some cases but not always, as you found yourself when you performed a libuvc installation.

Another fix is to downgrade the JetPack version from JP6 to 5.13, though I understand that this may not be possible for Ubuntu 22.04 users.

A third method for adding IMU support on JP6 and a USB cable connection is to install the ** RealSense MIPI driver** at the link below, though it is a complicated procedure.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Sometimes ROS topics will publish normally in RViz despite **control_transfer returned** warnings generating continuously in the log.  Is the IMU data viewable in RViz, please?  I note that you have the **Motion Module Force Pause** error in your log though, in which the streaming of the IMU data fails and cannot be recovered during that particular launch session.

************

In regard to installing pyrealsense2 with the **pip install pyrealsense2** method, this usually does not work with Jetson boards and it is recommended that the pyrealsense2 wrapper is compiled from source code on Jetson instead.

Does your Python script run if you launch it in sudo admin powers mode with `sudo python3 main_collision.py`

amaraashishmahatokushwaha, 2025-01-09:
Hi @MartyG-RealSense ,
Thank you for your faster response and well detailed explanation. Can you attached the link for installing pyrealsense2 wrapper  for ubuntu22.04 for jetson Oirn nano.  And while running using sudo python3 main_collision.py it shows error like there is no rclpy found or installed  and i have tried installing rclpy in root but i didn't achieve. so using this sudo seems doesn't works. and my last question is would you recommend to install MIPI driver using third method you suggest?
Thank you again.

MartyG-RealSense, 2025-01-09:
The **buildLibealsense.sh** build script provided at the JetSonHacks website may work well for you, as it builds the librealsense SDK with CUDA support on Jetson and compiles the pyrelsense2 wrapper at the same time.

https://github.com/JetsonHacksNano/installLibrealsense/tree/master

If you are able to install the MIPI driver then it is the optimal solution for providing IMU support on JetPack 6.  Due to the complexity of the installation procedure though, if you require advice about it then I recommend posting a question at the official support page for the driver.

https://github.com/IntelRealSense/realsense_mipi_platform_driver/issues

amaraashishmahatokushwaha, 2025-01-20:
Hi @MartyG-RealSense I tried with the link you provided for installing sdk with cuda support and compiles pyrealsense2 from jetson hacks,but after the completion of installation when i plug in the camera and open realsense it shows please insert the camera. no device found.

MartyG-RealSense, 2025-01-20:
Is the camera still able to be detected in realsense-viewer?  If it is, does the Motion Module now work?

Have you attempted to install the MIPI driver as suggested above, please?

amaraashishmahatokushwaha, 2025-01-21:
Hi @MartyG-RealSense ,
My first priority is to run .py file . And do i need to install MIPI driver for it?  The steps which i follow from jetson hacks it didn't work in my case,and then i again install sdk with libuvc.sh from official github page changing some make command you suggested in line no 46. and now realsense-viewer works. But motion modules still not working,when i toggle on in motion modules, it says no frame received and all tab were halted suddenly when i toggle motion modules on . For time being just suggest me and give me best possible way that i can run .py file which import pyrealsense2.

MartyG-RealSense, 2025-01-21:
At https://github.com/IntelRealSense/librealsense/issues/12807#issuecomment-2118169814 one of my Intel RealSense colleagues suggests to a Jetson Orin user with librealsense 2.55.1 and JetPack 6 whose camera was undetected by pyrealsense2 to configure the **udev** device handling rules by running the script `./scripts/setup_udev_rules.sh`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/setup_udev_rules.sh

closed 2025-02-13 by MartyG-RealSense