realsenseai/librealsense #13671: RealSense device detected only when using sudo

SwooshMihu, 2025-01-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Linux (Ubuntu 22) / Jetpack 6.1  | 
| Kernel Version (Linux Only)    |  5.15.148  | 
| Platform                                 | Jetson Orin Nano |
| SDK Version                            |  2.55.1  |
| Segment			|  Robotics  | 

### Issue Description
I am running in a peculiar issue with the realsense-viewer. I've installed the SDK using libuvc_installation.sh  because I'm using a Jetson Orin Nano running on Jetpack 6.1. Realsense-viewer doesn't find the camera without running the command with sudo privileges. 

Error log:
Found 0 RealSense devices (0xff requested & 0xff from device-mask in settings)

SwooshMihu, 2025-01-13:
Thank you for the answer. Can you be more specific about what "using sudo admin powers" mean and how that can be practically integrated in this application?

MartyG-RealSense, 2025-01-13:
The link below has information about sudo admin power on Ubuntu.

https://help.ubuntu.com/kubuntu/desktopguide/hu/root-and-sudo.html

using sudo just requires putting the word sudo in front of a program's launch command in the terminal, such as `sudo realsense-viewer`

SwooshMihu, 2025-01-13:
Yes, ok. I have stated in the issue definition that by using SUDO, the realsesnse-viewer works as expected. The issue arrises when i am working in a context that forbids me from using the admin privileges. 
1.Do you have another suggestion beyond that?
or 
2.Is this fix something on the development roadmap for the intel-realsense sdk?

MartyG-RealSense, 2025-01-20:
Hi @SwooshMihu  Do you require further assistance with this case, please?  Thanks!

SwooshMihu, 2025-01-20:
That solution did not work, i have found another fix which i will document shortly

MartyG-RealSense, 2025-01-21:
I'm pleased to hear that you found a fix.  Thanks so much for your intention to share the details of it!

MartyG-RealSense, 2025-02-04:
Hi @JPutwente  My understanding is that using the **sudo** admin powers command avoids the need for the udev device handing rules to be used.

For some Jetson users, reloading the udev files helps, like in the case at the link below.

https://forums.developer.nvidia.com/t/problem-installing-intel-d435-camera/252862/7

```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
```

MartyG-RealSense, 2025-02-04:
Are you installing librealsense into a ROS workspace, please?  If you are then it should be installed at a location outside of the workspace, otherwise it may not work.

JPutwente, 2025-02-04:
[image]

yes Ive placed it in the ros_ws. It seemed to build a ros_pkg and I thought this pkg was needed to build before the realsense2_camera pkg was build.

So should I just place it outside the ws, use the installation.sh. Then colcon build the ros wrapper and then it should work?

MartyG-RealSense, 2025-02-04:
Yes, place librealsense outside of the workspace and colcon-build the ROS wrapper inside the workspace.   This will increase the chance of it working.

JPutwente, 2025-02-04:
Before I'll try this and rebuild everything, I want to see if building from source is necessary.
Since the realsense-viewer did work in sudo, basically using the systemwide installment. 

I'll first clean my device with
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

and reinstall the sdk and wrapper with apt

`sudo apt install ros-jazzy-librealsense2*`
`sudo apt install ros-jazzy-realsense2-*`

If this doesnt work, then ill clean again and try the from source method.
This is probably going to take me 2 hours or so

MartyG-RealSense, 2025-02-04:
Okay, I look forward to your next report.  Good luck!

JPutwente, 2025-02-04:
Oke I also tried building from source with the sdk outside the ros_ws.
I encountered some issues regarding the user/local/bin files kept containing 2.55.1 files. I had to remove all of them manually. Then after building the installation folder installed again de 2.55.1 files instead of the 2.56.3

This is about the all the cmake & target files in /usr/local/lib/cmake/realsense2 and realsense2-gl
And all the files in /usr/local/lib/
`sudo cp -r ~/librealsense-2.56.3/build/Release/* /usr/local/lib/`

However this still resulted in:

`home/jetson/JP-ASAI_Add-on/asai-add-on-jetson-orin-sw-dev/ASAI_Add-on/src/realsense-ros-4.56.3/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::imu_callback(rs2::frame)’:
/home/jetson/JP-ASAI_Add-on/asai-add-on-jetson-orin-sw-dev/ASAI_Add-on/src/realsense-ros-4.56.3/realsense2_camera/src/base_realsense_node.cpp:519:71: error: ‘class rs2::motion_frame’ has no member named ‘get_combined_motion_data’
  519 |             auto combined_motion_data = frame.as<rs2::motion_frame>().get_combined_motion_data();
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/realsense2_camera.dir/build.make:93: CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/realsense2_camera.dir/build.make:121: CMakeFiles/realsense2_camera.dir/src/rs_node_setup.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/realsense2_camera.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< realsense2_camera [1min 17s, exited with code 2]

Summary: 3 packages finished [1min 19s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera`

but hey, this did resulted in not having to use `sudo` anymore.

JPutwente, 2025-02-04:
So its either Having the apt install where i can only run with `sudo`
Or have the source build which cant be `colcon build`

MartyG-RealSense, 2025-02-05:
Thanks very much for the additional information.  It's excellent to see that you made significant progress during your investigations.

Have you tried reloading the udev rules please, as suggested earlier at https://github.com/IntelRealSense/librealsense/issues/13671#issuecomment-2633869940

JPutwente, 2025-02-05:
Okey Good news!

Ive kind of made it work:

1. Download the sdk 2.56.3 and build from source with the cuda flag on
2. This generates for some reason 2.55.1 files in my /usr/local/ (maybe because of the flag?)
3. get the default ros2-master branch git clone
4. `sudo apt-get install ros-jazzy-diagnostic-updater`
5. Colcon build finishes! :D
6. run the launch
7. weird error:

`jetson@localhost:~/JP-ASAI_Add-on/asai-add-on-jetson-orin-sw-dev/ASAI_Add-on$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/jetson/.ros/log/2025-02-05-16-44-13-946028-localhost-50722
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [50738]
[realsense2_camera_node-1] [INFO] [1738773854.517069597] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1738773854.517347460] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1738773854.517390821] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1738773854.946654613] [camera.camera]: Device with serial number 239722071440 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1738773854.946907163] [camera.camera]: Device with physical ID 1-2.3-7 was found.
[realsense2_camera_node-1] [INFO] [1738773854.946954332] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1738773854.947553804] [camera.camera]: Device with port number 1-2.3 was found.
[realsense2_camera_node-1] [INFO] [1738773854.947620365] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1738773854.947673807] [camera.camera]: Device 239722071440 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1738773854.947988855] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1738773854.948978480] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1738773854.949025777] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1738773854.949060242] [camera.camera]: Device Serial No: 239722071440
[realsense2_camera_node-1] [INFO] [1738773854.949092307] [camera.camera]: Device physical port: 1-2.3-7
[realsense2_camera_node-1] [INFO] [1738773854.949127668] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1738773854.949158517] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1738773854.949190197] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1738773857.866862903] [camera.camera]: Set ROS param depth_module.depth_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1738773857.867947795] [camera.camera]: Set ROS param depth_module.infra_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1738773858.546443275] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1738773858.548503487] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1738773858.548735173] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1738773858.560498833] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1738773858.560749847] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1738773858.650095673] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1738773858.735827872] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1]  05/02 16:44:18,737 WARNING [281473340727360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1738773858.739561823] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  05/02 16:44:18,748 WARNING [281473340727360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1738773858.765544980] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1738773858.773424701] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  05/02 16:44:19,066 WARNING [281473340727360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

I'll have to check if this also work with just the standard sdk of the master branch
but if you know what this error is that would already help out a lot
Also could you maybe give me some insight why the sdk 2.56.3 would generate 2.55.1 binary and target files?

MartyG-RealSense, 2025-02-06:
A couple of causes of this **control_transfer returned** warning can be:

1.  A communication problem between the camera and the computer, typically related to a problem with the USB port or USB cable; or

2.  Data is arriving at the USB port faster than the computer's CPU can process it. Reducing the resolution or FPS speed of the RealSense streams may reduce the number of control_transfer warnings that are published in the log.

Sometimes topics will be published successfully in RViz even if control_transfer warnings are generating continuously in the log.

I note that your camera connection is being detected as slow USB 2.1 instead of fast USB 3.2.  If your camera is plugged into a USB 3 port on the Jetson, please try unplugging the micro-sized end of the USB cable from the side of the camera, turning the connector around the other way and re-inserting it to see whether the camera is then detected as USB 3.2.

In regard to why you get 2.55.1 binaries from a 2.56.3 beta installation, I do not know the reason for that occurring.

JPutwente, 2025-02-06:
@MartyG-RealSense So i was indeed using a wrong USB cable, oops. But it wasnt the problem.

I've Reinstalled the Jetson with a new image just to be sure I didnt have any cross installation.
I also now have builded bout the sdk and wrapper from source, both from the master branch.

how ever I still receive this problem.

`jetson@localhost:~/JP_ASAI_Add-on/asai-add-on-jetson-orin-sw-dev/ASAI_Add-on$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1738843405.702760749] [camera.camera]: RealSense ROS v4.55.1
[INFO] [1738843405.702941617] [camera.camera]: Built with LibRealSense v2.55.1
[INFO] [1738843405.702986482] [camera.camera]: Running with LibRealSense v2.55.1
[INFO] [1738843406.405597786] [camera.camera]: Device with serial number 239722071440 was found.

[INFO] [1738843406.405862016] [camera.camera]: Device with physical ID 2-1.3-7 was found.
[INFO] [1738843406.405905921] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1738843406.406341067] [camera.camera]: Device with port number 2-1.3 was found.
[INFO] [1738843406.406391340] [camera.camera]: Device USB type: 3.2
[INFO] [1738843406.406617969] [camera.camera]: getParameters...
[INFO] [1738843406.407438435] [camera.camera]: JSON file is not provided
[INFO] [1738843406.407489412] [camera.camera]: Device Name: Intel RealSense D435I
[INFO] [1738843406.407520901] [camera.camera]: Device Serial No: 239722071440
[INFO] [1738843406.407550277] [camera.camera]: Device physical port: 2-1.3-7
[INFO] [1738843406.407575494] [camera.camera]: Device FW version: 5.16.0.1
[INFO] [1738843406.407600614] [camera.camera]: Device Product ID: 0x0B3A
[INFO] [1738843406.407627079] [camera.camera]: Sync Mode: Off
[INFO] [1738843410.653944278] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1738843410.654169947] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1738843410.654255741] [camera.camera]: Stopping Sensor: Motion Module
[INFO] [1738843410.859074845] [camera.camera]: Starting Sensor: Depth Module
[INFO] [1738843410.989989804] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1738843410.990149583] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1738843410.990190832] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
 06/02 12:03:30,992 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[INFO] [1738843410.992794345] [camera.camera]: Starting Sensor: RGB Camera
 06/02 12:03:31,003 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[INFO] [1738843411.025155661] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
 06/02 12:03:31,030 WARNING [281472963436608] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [1738843411.036130109] [camera.camera]: Starting Sensor: Motion Module
[INFO] [1738843411.052802314] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[INFO] [1738843411.052957517] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [1738843411.058493190] [camera.camera]: RealSense Node Is Up!
 06/02 12:03:31,119 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

Then this just repeats itself:
` 06/02 12:03:31,119 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

MartyG-RealSense, 2025-02-06:
Thanks very much for the update!

1.  Are the topics published and viewable in RViz despite the warnings?

2.  Do the warnings still occur if you launch with **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node`

3.  Do the warnings still occur if you add `initial_reset:=true` to the ros2 launch instruction to reset the camera at launch?

JPutwente, 2025-02-06:
Thank you for the tips!

1: Yes everything is visable with the ros2 topic list and in RViz however nothing is being sent/received.
2: same error, The example I showed before was with ros2 run.
3: Also didnt make a difference

MartyG-RealSense, 2025-02-06:
Let's try ensuring that most of the streams that are opened in the log are disabled so that only depth is enabled.

`ros2 launch realsense2_camera rs_launch.py enable_color:=false enable_infra1:=false enable_infra2:=false enable_gyro:=false enable_accel:=false`

If the warnings do not occur or they occur much less frequently, then next try having depth + color are enabled.  This will tell us whether the color stream is the cause of the warnings.

`ros2 launch realsense2_camera rs_launch.py enable_infra1:=false enable_infra2:=false enable_gyro:=false enable_accel:=false`

JPutwente, 2025-02-07:
Sadly makes no difference than with just:
`ros2 launch realsense2_camera rs_launch.py`

Ive also managed to update the sdk and the wrapper to 2.56.0 to see if it was a ros-jazzy issue.
But this is not different from before.

MartyG-RealSense, 2025-02-07:
I note that you have the Orin Nano model of Jetson.  Intel strongly recommend enabling the **barrel jack** connector on Nano boards for extra power using the instructions at the link below.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

JPutwente, 2025-02-07:
I have already been working the Jetson Orin Nano with the barrel jack connector as its power source.
After being stuck with this now for a week I have come to the conclusion to go back to Jetpack with ubuntu 22. 
Hopefully this can just be used with to released master branches of the sdk and the ros wrapper.

Thank you for all your help and effort!

MartyG-RealSense, 2025-02-08:
You are very welcome.  Please do update me later about your progress with Ubuntu 22.  Good luck!

closed 2025-02-28 by MartyG-RealSense

MartyG-RealSense, 2025-03-03:
It's great to hear that you achieved a successful solution - thanks very much for the update!  :)

Ziller0831, 2025-04-03:
Hello, I got the same problem.

| Info                          |                               |
| ----------------------------- | ----------------------------- |
| Camera Model                  | D455f                 |
| Librealsense SDK Version      | LibRealSense v2.55.1          |
| Firmware Version              | 5.15.1.55                      |
| Operating System              | JetPack6.0 rev2(Ubuntu 22.04) |
| Platform                      | Jetson Orin NX                |
| ROS Distro                    | Humble                        |
| RealSense ROS Wrapper Version | v4.55.1                       |

## Issue Description
Last week, I was still able to use RealSense ROS without any issues. However, as of yesterday, when I try to run `ros2 launch realsense2_camera rs_launch.py`, I get the following message:

```bash
ced@orin-nx-recomputer-mini:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py                                             
[INFO] [launch]: All log files can be found below /home/ced/.ros/log/2025-04-03-16-11-01-655996-orin-nx-recomputer-mini-507618
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rs-ros-wrapper-1]: process started with pid [507619]
[rs-ros-wrapper-1] [INFO] [1743667862.094125430] [camera.camera]: RealSense ROS v4.55.1
[rs-ros-wrapper-1] [INFO] [1743667862.094434974] [camera.camera]: Built with LibRealSense v2.55.1
[rs-ros-wrapper-1] [INFO] [1743667862.094503360] [camera.camera]: Running with LibRealSense v2.55.1
[rs-ros-wrapper-1] [WARN] [1743667862.111290703] [camera.camera]: No RealSense devices were found!
```

When I use `lsusb`, I can see information about the D455f:

```bash
ced@orin-nx-recomputer-mini:~/ros2_ws$ lsusb
Bus 002 Device 006: ID 8086:0b5c Intel Corp. Intel(R) RealSense(TM) Depth Camera 455f
```

However, when I run `rs-enumerate-devices`, it shows `No device detected. Is it plugged in?`. But if I run `rs-enumerate-devices` with `sudo`, the information displays correctly:

```bash
ced@orin-nx-recomputer-mini:~/ros2_ws$ sudo rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense D455F
    Serial Number                 :     338122301450
    Firmware Version              :     5.15.1.55
    Recommended Firmware Version  :     5.16.0.1
    Physical Port                 :     2-3-6
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B5C
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     250143060606
    Firmware Update Id            :     250143060606
    Dfu Device Path               :

Stream Profiles supported by Stereo Module
 (Omitted here for brevity…)

Stream Profiles supported by RGB Camera
 (Omitted here for brevity…)

Stream Profiles supported by Motion Module
 (Omitted here for brevity…)
```
This has caused me a lot of trouble when using RealSense ROS.

JPutwente, 2025-04-03:
When this happened to me it usually meant the libsense was installed with an apt pkg while also having the source code installed. I’d remove the apt pkg and keep the sdk libsense build from source (maybe build it again after the remove)

MartyG-RealSense, 2025-04-03:
Thanks so much @JPutwente for your advice to @Ziller0831

Ziller0831, 2025-04-07:
@JPutwente It's work!!!, Thank you very much for the advice.

JasonTan94, 2025-06-04:
That's some hardest setup I've done. Came here to record down for my future self and also anyone who needs it.

| Item  |  Info |
| ------------- | ------------- |
| Camera Model | D435i  |
Librealsense SDK Version | LibRealSense v2.55.1
Platform | Jetson AGX Orin Dev Kit
Operating System | Ubuntu 22.04
L4T | L4T 36.3.0 JetPack 6.0
Kernel Version | 5.15.136-rt-tegra
ROS Distro | Humble

Some conclusions here:
If you found `rs-enumerate-devices` can only work with `sudo`

> When this happened to me it usually meant the libsense was installed with an apt pkg while also having the source code installed. I’d remove the apt pkg and keep the sdk libsense build from source (maybe build it again after the remove)

Is the way to fix it. Remove all `librealsense` from both places and make a clean installation from source again.
```
sudo apt purge librealsense2*
dpkg -l | grep librealsense
```

After that, if lots of 
```
(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
errors when running ros2, just like what can be seen from 

> I've Reinstalled the Jetson with a new image just to be sure I didnt have any cross installation. I also now have builded bout the sdk and wrapper from source, both from the master branch.
> 
> how ever I still receive this problem.
> 
> jetson@localhost$ ros2 run realsense2_camera realsense2_camera_node 
> 
> [INFO] [1738843405.702760749] [camera.camera]: RealSense ROS v4.55.1
> [INFO] [1738843405.702941617] [camera.camera]: Built with LibRealSense v2.55.1
> [INFO] [1738843405.702986482] [camera.camera]: Running with LibRealSense v2.55.1
> [INFO] [1738843406.405597786] [camera.camera]: Device with serial number 239722071440 was found.
> 
> [INFO] [1738843406.405862016] [camera.camera]: Device with physical ID 2-1.3-7 was found.
> [INFO] [1738843406.405905921] [camera.camera]: Device with name Intel RealSense D435I was found.
> [INFO] [1738843406.406341067] [camera.camera]: Device with port number 2-1.3 was found.
> [INFO] [1738843406.406391340] [camera.camera]: Device USB type: 3.2
> [INFO] [1738843406.407489412] [camera.camera]: Device Name: Intel RealSense D435I
> [INFO] [1738843406.407520901] [camera.camera]: Device Serial No: 239722071440
> [INFO] [1738843406.407550277] [camera.camera]: Device physical port: 2-1.3-7
> [INFO] [1738843406.407575494] [camera.camera]: Device FW version: 5.16.0.1
> [INFO] [1738843406.407600614] [camera.camera]: Device Product ID: 0x0B3A
> [INFO] [1738843410.653944278] [camera.camera]: Stopping Sensor: Depth Module
> [INFO] [1738843410.654169947] [camera.camera]: Stopping Sensor: RGB Camera
> [INFO] [1738843410.654255741] [camera.camera]: Stopping Sensor: Motion Module
> [INFO] [1738843410.859074845] [camera.camera]: Starting Sensor: Depth Module
> [INFO] [1738843410.989989804] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
> [INFO] [1738843410.990149583] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
> [INFO] [1738843410.990190832] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
> 06/02 12:03:30,992 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error:  > Resource temporarily unavailable, number: 11
> 06/02 12:03:31,003 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error:  > Resource temporarily unavailable, number: 11
> [INFO] [1738843411.025155661] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
> [INFO] [1738843411.036130109] [camera.camera]: Starting Sensor: Motion Module
> [INFO] [1738843411.052802314] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
> [INFO] [1738843411.052957517] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
> [INFO] [1738843411.058493190] [camera.camera]: RealSense Node Is Up!
06/02 12:03:31,119 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`
> 
> Then this just repeats itself: ` 06/02 12:03:31,119 WARNING [281472938074176] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

Then, it's something that will happen on _Jetpack 6.0_ onwards, where you need to modify the kernel to make it work.
Please follow these repos
https://github.com/jetsonhacks/jetson-orin-kernel-builder
https://github.com/jetsonhacks/jetson-orin-librealsense
It will guide you on enabling the kernel drivers, find both YouTube vids and follow along. They will make your life whole lot easier.

MartyG-RealSense, 2025-06-04:
Thanks so much @JasonTan94 for sharing your hard-won setup knowledge with the RealSense community!

shrutic12, 2026-02-27:
> Hi Marty after a lot of time trying to work with Isaac docker (which also did not really work). I decided to just dive into the sdk code and just make a super simple wrapper for ROS2. The code I mainly used was the rs-pointcloud.cpp from the sdk examples. This resulted in getting a pointcloud into ROS2 :D

Hello @JPutwente ! I am facing some strange issues with the realsense ROS wrapper where enabling the "align_depth" parameter causes depth frames to not be published at all. Even after disabling alignement, I have frames published at maximim of 5 fps for depth and 2 fps for colour. 
I am investigating rewriting the ROS2 wrapper to improve performance. I see you have done the same, would you be able to share any code? 

Thanks